area515/Photonic3D #4: GUI reorganization

ergobot, 2015-01-21:
Current UI sections (bootstrap divs) are functional but do not guide the user due to page placement.  Organize UI sections to support intuitive use.

WesGilster, 2015-02-06:
Rewrite to Bootcards?  :)

ergobot, 2015-02-06:
Yes, its neatly organized and touch/mobile friendly

http://demo.bootcards.org/

WesGilster, 2015-02-09:
Throwing out some ideas for the dashboard:
Manage Printers
Manage Jobs
Manage Host
Manage Uploads
Printer Control
    DLP Control
    GCode Control

kloknibor, 2015-02-09:
maybe something like this?
http://imgur.com/qwNvrvI

kloknibor, 2015-02-19:
maybe a nice function would be to make it possible to add a button or an extra entry in the menu which could start a custom script (This would be great to use in combination with WiringPi this way you could power down your projecter or even the whole printer when unused or when done with the job when you're not home end you could also turn it on all on :) (as long as the pi still has power))

WesGilster, 2015-02-20:
I'm planning on supporting the shutdown projector natively with plugable commands you can send to the project. It would be a very quick change to do that. This still might be a cool function to have though.

WesGilster, 2015-03-23:
So even though print on the fly from STL has been implemented and working pretty nicely, it's not easy(in fact it's pretty tough) to add your own customized printers right now. In order to build a printer that is capable of STL printing, several calculators(javascript), gcode (FreeMarker templates) and variables need to be setup in the machine/profile xml. The GUI rewrite will make all of this a snap to setup. It will also allow migrating machines and slicing profiles from Creation Workshop a little easier as well. The migration won't be perfect since Creation Workshop uses an proprietary templating system for GCode and we'll be using FreeMarker for templating.

SouthPaw-42, 2015-07-16:
Add more manual controls. I.E. Vat Peal up / Down, Projector Shutter Open/Closed, Projector On/Off, Panic Shop button (For failed prints).

Fix Firefox Display issues.

sbryan, 2015-07-16:
FYI, I've started adding some of the projector controls (power on/off, shutter open/close), but, as noted in my branch, I'm sure the current way I am doing it is a complete hack... should have a more dynamic way to display them after detection that a printer exists and the model is supported...

The change can be seen in its current form at: https://github.com/sbryan/Creation-Workshop-Host/commit/309eb8c87a59ce676ecb9726736d745a8fd1a51c

I've not submitted a PR yet, as I'd really like to figure out the "proper" way to do this

WesGilster, 2015-07-17:
The way you did it will work just fine if you just want it to work, but this is the direction I would take:
               A. Add restful method openShutter(PrinterName) to org.area515.resinprinter.services.MachineService to be executed by javascript.
               B. Add new Xml gCode methods get/setgCodeOpenShutter() in org.area515.resinprinter.printer.SlicingProfile
               C. Add new method openShutter() to org.area515.resinprinter.printer.Printer to be executed by the new method you created in the MachineService and utilize gcode from org.area515.resinprinter.printer.SlicingProfile
	       D. Don't forget to perform the actual shutter operations at the appropriate times inside of the org.area515.resinprinter.job.STLFileProcessor like this:
if (slicingProfile.getgCodeOpenShutter() != null && slicingProfile.getgCodeOpenShutter().trim().length() > 0) {
				printer.getGCodeControl().executeGCodeWithTemplating(printJob, slicingProfile.getgCodeOpenShutter());
			}

The above If statement is the closest equivalent of an "isSupported()" method that we have. At the moment, the only other way that we detect if an option is supported is by calling it. If we get an InappropriateDeviceException with the text:
org.area515.resinprinter.printer.MachineConfig.NOT_CAPABLE
then the printer doesn't support the option.

WesGilster, 2015-08-12:
I'm going to start a pretty big restful refactoring to support the GUI rewrite. Mostly just reorganization, renaming, bundling like functionality in the services and there will be a clear delineation between accessing jobs and files through the API. There will be enough changes that, I'm pretty much thinking I'll just dump the old gui and not migrate it.

Fleshing out the bootcards a bit:   
A. Printers:   
   1. Add/Change   
      i. Create Simple printer (No need to setup anything)   
      ii. Create Advanced printer   
         a. Register printer with notifiers   
         b. Register printer with advertisers   
         c. Create javascript masks   
         d. Create variable calculators  
         e. Camera setup   
         f. Setup printer variables and settings.   
   2. View currentJob status   
   3. Delete   
B. Jobs   
   1. Pause/Start and Stop   
   2. View status   
   3. Control job   
      i. DLP Controls   
      ii. GCode Controls   
C. User Managment Setup   
   1. Setup user   
      i. username   
      ii. first/last name   
      iii. email   
   2. SSL TLS Authentication/Authorization   
D. Printable Files   
   1. Upload   
   2. Delete

kloknibor, 2015-08-12:
Sounds great! One thing I miss is saving on the raspberry pi so if we printed a file once we just click reprint or something ;)! And maybe it would be nice to see a compact version of current jub status at all time but that's not really nescassery

WesGilster, 2015-08-12:
Yeah, I was thinking the "Compact job status" would simply be placed on printer so you could see it at a glance.

kloknibor, 2015-08-12:
That's nice enough I guess ;)! And what about saving files to print on the rpi? 

What's your opinion on that?

WesGilster, 2015-08-12:
I'm not sure what you mean. Once you upload the file, it's saved on the Raspberry Pi. If at any point you want to print again, just hit the 'Start Job' button. You never have to upload the file file again (unless you delete the file)

kloknibor, 2015-08-13:
Hmm could it be that in older versions the files were deleted once rebooted? But you what you describe is what I mean ;) So you already have that function :p

WesGilster, 2015-08-13:
I can't really remember a time when we didn't have that feature.

WesGilster, 2015-08-15:
First screen checked in, no links are bound properly. All that that is bound to the backend is the number of printables and printers. Really it's barely enough to give you a taste of what is to come. Shameless copy of http://demo.bootcards.org/.

WesGilster, 2015-08-15:
To see the new gui set this variable in your config.properties:
hostGUI=resourcesnew
To see the old gui set this variable in your config.properties
hostGUI=resources

WesGilster, 2015-08-15:
Let me know how it looks from a Mobile standpoint

kloknibor, 2016-01-08:
Mobile it looks fine just unresponsive on mobile sometimes. On the desktop when looking at the currently printing screen it looks a bit like a mess... Could be better organised maybe in 2 seperate columns ;)!

WesGilster, 2016-01-18:
I just saw this comment. Yeah, the boot cards aren't arranged pretty on a desktop, I think it actually looks better on mobile. The proper bootstrap column classes need to be sprinkled throughout the app. I'd take pull requests from anyone who wants to take a crack at it.

kloknibor, 2016-03-07:
I think this one can be closed... Only problem left is the dray targets on mobile left

closed 2016-03-07 by WesGilster