area515/Photonic3D #5: upnp testing

ergobot, 2015-01-21:
Create test scenario for upnp.

WesGilster, 2015-01-30:
This enhancement is complete, in build (cwh-0.62.zip) 
Instructions for starting are in installnotes.txt
Need help testing.

kloknibor, 2015-02-04:
I installed biold 69 on my raspberry pi and tried to run the .bat file on me windows laptop. it does nothing but says in the terminal the following : 

C:\user\etc\etc\etc> -cp lib/*;. org.area515.resinprinter.client.Main
Error: Could not find or load main class org.area515.resinprinter.client.Main

I'm guess I did something wrong... the only thing on my windows pc is the .bat file... should I mirror the github repo on my windows laptop?

thanks,

robin

WesGilster, 2015-02-04:
It doesn't look like you performed the steps in the installnotes here:
https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/installnotes.txt

But on the other hand, it looks like the installation notes on area515 are wrong so that could be part of the problem. It seems as if that file didn't get properly merged into area515 repo.

Anyway, you need to unzip the https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/cwhClient-0.69.zip before you can double click on the browseprinter.bat.

kloknibor, 2015-02-05:
So I tested today again first it stayed at the following screen for some minutes : 
http://imgur.com/2X8Xs0U
but well that could be fine, but afterwards it went further and I got this : 
http://imgur.com/1aKlmaO 

and 

http://imgur.com/IYwCKrz

I looked up the ip 192.168.1.26 on my network and it's a bose sound system... I assume the way the .bat file works is that you put a server up on the pc and the creation workshop host should react to this but the bose sounds system is faster? the raspberry pi was btw on 192.168.1.30

one suggestion, would bonjour not be a much simpler and easier solution? there's a default deamon fot it and the plus is you can just use a simple name and you can even change names if you want several printers in a network, like cwh1.local and cwh2.local etc that would be easier than the batch file I guess... here is some more information : http://www.howtogeek.com/167190/how-and-why-to-assign-the-.local-domain-to-your-raspberry-pi/

hope this helped :)

WesGilster, 2015-02-05:
Yes, the stack traces help a ton. I didn't spend much time on that client and my error handling was pretty sloppy. It turns out that it probably found your printer but it had an error closing the UPNP connections and threw that exception rather than browse to your printer.

We initially contemplated all sorts of Broadcast services like WS-Discovery and BonJour, but settled on UPNP because it's been around, it's had widespread adoption on multiple platforms, lot's of functionality and was very easy to implement. 

We also figured people would want other forms of attempting to find their printer, so I made this mechanism pluggable. All that needs to be done is implement two easy methods:
public interface Advertiser {
	public void start(URI uri);
	public void stop();
}
Then modify the config.properties to specify an additional new advertiser class:
advertise.BonJourAdvertiser=true

For example, you could make an advertiser that emails the user where the URL is located, very flexible. You get the idea...

When we get to the point of updating the GUI, all of these advertisement options will be available from there, but that's for another day.

Fixed in (Build72)

kloknibor, 2015-02-05:
It's great to hear you use a pluggable mechanism :) and that we probably can choose later on :) 

i tested the new build today... it worked eventually... but got some glitches... what I did : 

1. I enabled ssl and browsed to the printer
2. login screen came up and I filled in a wrong pas, the thin that happend was that the login screen just disappeard and came back blank (talking about milliseconds) so no error message
3. logged in with the correct password and got the following error "You logged in wrong. Please delete your cookies and try again." that's a nice error ;)
4. I deleted the cookies, as soon as I reached the printer page again it was logged in... So I didn;'t have to fill in any data at all to come in.

conclusion, it works.... but well its kinda... how to call it nicely... it just can be a bit better... but as concept it works :)!

kloknibor, 2015-02-05:
I also just tried the UPNP service again.. didn;t work here is the cmd file : 

C:\Users\Robin\Downloads\cwh-0.72>java -cp lib/*;. org.area515.resinprinter.clie
nt.Main
feb 05, 2015 11:25:43 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: >>> Starting UPnP service...
feb 05, 2015 11:25:43 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: Using configuration: org.fourthline.cling.DefaultUpnpServiceConfiguration
feb 05, 2015 11:25:43 PM org.fourthline.cling.transport.RouterImpl <init>
INFO: Creating Router: org.fourthline.cling.transport.RouterImpl
feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Intel
(R) Centrino(R) Wireless-N 2230
feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Virtu
alBox Host-Only Ethernet Adapter
feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.StreamServerImpl in
it
INFO: Created server (for receiving TCP streams) on: /192.168.1.28:52042
feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.DatagramIOImpl init

INFO: Creating bound socket (for datagram input/output) on: /192.168.1.28
feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.StreamServerImpl in
it
INFO: Created server (for receiving TCP streams) on: /192.168.56.1:52045
feb 05, 2015 11:27:12 PM org.fourthline.cling.transport.impl.DatagramIOImpl init

INFO: Creating bound socket (for datagram input/output) on: /192.168.56.1
feb 05, 2015 11:27:12 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: <<< UPnP service started successfully
feb 05, 2015 11:27:13 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 05, 2015 11:27:13 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
feb 05, 2015 11:27:13 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 05, 2015 11:27:13 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
feb 05, 2015 11:27:14 PM org.fourthline.cling.UpnpServiceImpl$1 run
INFO: >>> Shutting down UPnP service...
java.lang.RuntimeException: Router error on shutdown: org.fourthline.cling.trans
port.RouterException: Router wasn't available exclusively after waiting 6000ms,
lock failed: WriteLock
        at org.fourthline.cling.UpnpServiceImpl.shutdownRouter(UpnpServiceImpl.j
ava:168)
        at org.fourthline.cling.UpnpServiceImpl$1.run(UpnpServiceImpl.java:143)
        at org.fourthline.cling.UpnpServiceImpl.shutdown(UpnpServiceImpl.java:15
2)
        at org.fourthline.cling.UpnpServiceImpl.shutdown(UpnpServiceImpl.java:13
4)
[... 78 lines cut ...]
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
feb 05, 2015 11:29:38 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 05, 2015 11:29:38 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
feb 05, 2015 11:29:40 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172
feb 05, 2015 11:30:08 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 05, 2015 11:30:08 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true

WesGilster, 2015-02-05:
Wow you have a ton of gadgets on your network, and they look to be misbehaving a bit. This turned out to be a really good test, because it runs flawlessly on my network.

Looks like all of those gadgets tie up the router and cause some problems on the router shutdown. I accidentally had the logic reversed in the new error handling. Fixed in (build75)

kloknibor, 2015-02-10:
tested today with build 79 and it still didn't work here's the log from my ssh client : 

C:\Users\Robin\Desktop\cwh host>java -cp lib/*;. org.area515.resinprinter.client
.Main
feb 10, 2015 12:07:30 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: >>> Starting UPnP service...
feb 10, 2015 12:07:30 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: Using configuration: org.fourthline.cling.DefaultUpnpServiceConfiguration
feb 10, 2015 12:07:30 PM org.fourthline.cling.transport.RouterImpl <init>
INFO: Creating Router: org.fourthline.cling.transport.RouterImpl
feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Intel
(R) Centrino(R) Wireless-N 2230
feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Virtu
alBox Host-Only Ethernet Adapter
feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.StreamServerImpl in
it
INFO: Created server (for receiving TCP streams) on: /192.168.1.28:62273
feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.DatagramIOImpl init

INFO: Creating bound socket (for datagram input/output) on: /192.168.1.28
feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.StreamServerImpl in
it
INFO: Created server (for receiving TCP streams) on: /192.168.56.1:62276
feb 10, 2015 12:08:58 PM org.fourthline.cling.transport.impl.DatagramIOImpl init

INFO: Creating bound socket (for datagram input/output) on: /192.168.56.1
feb 10, 2015 12:08:58 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: <<< UPnP service started successfully
feb 10, 2015 12:08:59 PM org.fourthline.cling.UpnpServiceImpl$1 run
INFO: >>> Shutting down UPnP service...
Exception in thread "main" java.lang.RuntimeException: Router error on shutdown:
 org.fourthline.cling.transport.RouterException: Router wasn't available exclusi
vely after waiting 6000ms, lock failed: WriteLock
        at org.fourthline.cling.UpnpServiceImpl.shutdownRouter(UpnpServiceImpl.j
ava:168)
        at org.fourthline.cling.UpnpServiceImpl$1.run(UpnpServiceImpl.java:143)
        at org.fourthline.cling.UpnpServiceImpl.shutdown(UpnpServiceImpl.java:15
2)
        at org.fourthline.cling.UpnpServiceImpl.shutdown(UpnpServiceImpl.java:13
4)
        at org.area515.resinprinter.client.Main.main(Main.java:33)
Caused by: org.fourthline.cling.transport.RouterException: Router wasn't availab
le exclusively after waiting 6000ms, lock failed: WriteLock
        at org.fourthline.cling.transport.RouterImpl.lock(RouterImpl.java:497)
        at org.fourthline.cling.transport.RouterImpl.lock(RouterImpl.java:510)
        at org.fourthline.cling.transport.RouterImpl.disable(RouterImpl.java:158
)
        at org.fourthline.cling.transport.RouterImpl.shutdown(RouterImpl.java:19
9)
        at org.fourthline.cling.UpnpServiceImpl.shutdownRouter(UpnpServiceImpl.j
ava:162)
        ... 4 more
feb 10, 2015 12:09:05 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 10, 2015 12:09:05 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
[... 97 lines cut ...]
feb 10, 2015 12:11:50 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172
feb 10, 2015 12:11:53 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172
feb 10, 2015 12:12:15 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 10, 2015 12:12:15 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
feb 10, 2015 12:12:22 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172
feb 10, 2015 12:12:24 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
 describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172

WesGilster, 2015-02-11:
It doesn't look like you aren't using the correct version of the client 
based on this line:
at org.area515.resinprinter.client.Main.main(Main.java:33)

Line 33 is an invalid line in the current codebase (build 79).

Wes G.

kloknibor, 2015-02-17:
So I  tried it again today, downloaded everything fresh from github and get A bit different log but it sill doesn't work... here it is : 

C:\Users\Robin\Downloads\cwh-0.79>java -cp lib/*;. org.area515.resinprinter.clie
nt.Main
feb 17, 2015 2:47:59 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: >>> Starting UPnP service...
feb 17, 2015 2:47:59 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: Using configuration: org.fourthline.cling.DefaultUpnpServiceConfiguration
feb 17, 2015 2:47:59 PM org.fourthline.cling.transport.RouterImpl <init>
INFO: Creating Router: org.fourthline.cling.transport.RouterImpl
feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.MulticastReceiverImp
l init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.MulticastReceiverImp
l init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Intel
(R) Centrino(R) Wireless-N 2230
feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.MulticastReceiverImp
l init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.MulticastReceiverImp
l init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Virtu
alBox Host-Only Ethernet Adapter
feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.StreamServerImpl ini
t
INFO: Created server (for receiving TCP streams) on: /192.168.1.28:64359
feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.DatagramIOImpl init
INFO: Creating bound socket (for datagram input/output) on: /192.168.1.28
feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.StreamServerImpl ini
t
INFO: Created server (for receiving TCP streams) on: /192.168.56.1:64363
feb 17, 2015 2:49:23 PM org.fourthline.cling.transport.impl.DatagramIOImpl init
INFO: Creating bound socket (for datagram input/output) on: /192.168.56.1
feb 17, 2015 2:49:23 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: <<< UPnP service started successfully
feb 17, 2015 2:49:24 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 17, 2015 2:49:24 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
feb 17, 2015 2:49:25 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172
feb 17, 2015 2:49:26 PM org.fourthline.cling.UpnpServiceImpl$1 run
INFO: >>> Shutting down UPnP service...
feb 17, 2015 2:49:32 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172
feb 17, 2015 2:49:37 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 17, 2015 2:49:37 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
feb 17, 2015 2:50:02 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describe
WARNING: Device descriptor retrieval failed, no response: http://192.168.1.26:80
91/XD/BO5EBO5E-F00D-F00D-FEED-08DF1F13D172
feb 17, 2015 2:50:03 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describeService
WARNING: Received empty service descriptor:http://192.168.1.10:5000/ssdp/dummy.x
ml
feb 17, 2015 2:50:04 PM org.fourthline.cling.protocol.RetrieveRemoteDescriptors
describe
WARNING: Device service description failed: (RemoteDevice) Identity: (RemoteDevi
ceIdentity) UDN: uuid:73796E6F-6473-6D00-0000-00113205b671, Descriptor: http://1
92.168.1.10:5000/ssdp/desc-DSM-eth0.xml, Root: true
Batchopdracht beëindigen (J/N)?

The last line ("Batchopdracht beëindigen (J/N)?") means : "end Batchtask (Y/N)?"

WesGilster, 2015-02-18:
Actually, this log is different, notice that there is no longer a stack trace. I also notice that it has also initiated shutdown of upnp in a little over 2 seconds:
eb 17, 2015 2:47:59 PM org.fourthline.cling.UpnpServiceImpl 
INFO: >>> Starting UPnP service...
feb 17, 2015 2:49:26 PM org.fourthline.cling.UpnpServiceImpl$1 run
INFO: >>> Shutting down UPnP service...

I will only shutdown upnp for 1 of 2 reasons:
1. I found the printer
2. I haven't found the printer after 7 seconds.

All of this means I'm sure it's finding your printer. The problem is that upnp is still not shutting down properly. In order for upnp to shutdown properly, it needs to wait until all responses have arrived back from each of the devices it was communicating with. From what I've seen on your network, I can't rely on those devices to respond in a reasonable amount of time. My new plan is to cut all of this short and forget a clean shutdown entirely. The long and the short of this is that I've changed the upnp search to be asynchronous now. Fixed in (build 70)

kloknibor, 2015-02-27:
It just can't find my printer it tells me now... here's what I get : 

http://i.imgur.com/j1cMzqQ.png

WesGilster, 2015-02-27:
Was it showing that message all along? Are you sure your Raspberry Pi isn't on a different subnet?
Could you run the following on your Raspberry Pi:
ifconfig

and then run this on the windows computer that is searching for the raspberry pi:
ipconfig

kloknibor, 2015-02-27:
I will tonight ;) I've got only 1 raspberry pi and I still switch between octopi and CWH host (someone asked a print on my old printer) but the only difference I could think of is that my laptop is on wifi and the pi is connected by LAN. I don't have any sub nets, I only use 2,4GHZ and haven't any repeaters installed...

WesGilster, 2015-02-27:
Yep, that's why mine doesn't work. My wifi is 10.0.0.x and my wired is 192.168.0.x. Multicast between subnets has to be setup in your router. I wish that worked by default...

Hey that's good news about the new pi. I'm still finishing up the slicer, there are some annoyances with the polygon colorization I have to work out on complex 3d models. I was hoping to have something working this weekend.

WesGilster, 2015-03-23:
Still waiting for ip results from the computer you are testing on and the ip of your raspberry Pi.

kloknibor, 2015-04-29:
Sorry for not responding so long... my lan cables were broken and I didn;t have any time to fix that for the pi... but it's all up and running now ;)

these are from the raspberry pi : 
eth0      Link encap:Ethernet  HWaddr b8:27:eb:7a:ec:3b
          inet addr:192.168.1.32  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1239 errors:0 dropped:1 overruns:0 frame:0
          TX packets:666 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:292570 (285.7 KiB)  TX bytes:335140 (327.2 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:27 errors:0 dropped:0 overruns:0 frame:0
          TX packets:27 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8395 (8.1 KiB)  TX bytes:8395 (8.1 KiB)

ipconfig on my pc
U in do ws IP Co n f igurat io n
Wireless LAN adapter Wireless Network Connection 3:
Media State ..................... : Media disconnected
Connection-specific DNS Suffix . :
Wireless LAN adapter Wireless Network Connection 2:
Media State ..................... : Media disconnected
Connection-specific DNS Suffix . :
Wireless LAN adapter Wireless Network Connection:
Connection-specific DNS Suffix . :
Link-local IP06 Address ......... : fe80::llc2:338f:ae55:742axl3
IPu4 Address.......................:	192.168.1.20
Subnet Mask ...................... :	255.255.255.0
Default Gateway .................. :	192.168.1.1
Ethernet adapter Bluetooth Network Connection:
Media State ..................... : Media disconnected
Connection-specific DNS Suffix . :
Tunnel adapter isatap.<3253A640-1985-4C59-99A4-623CA165960C>:
Media State ..................... : Media disconnected
Connection-specific DNS Suffix . :
Tunnel adapter isatap.<48C34269-FB4A-4BF2-B30F-B101E78F7493>:
Media State ...................... :	Media disconnected
Connection-specific DNS Suffix . :
Tunnel adapter isatap.<2136EF05-03D5-4A95-92E9-6116AA64BElD>:
Media State ..................... : Media disconnected
Connection-specific DNS Suffix . :
Tunnel adapter isatap.<2BC64D76-A047-4A96-A065-DBCE286D8826>:
Media State ..................... : Media disconnected
Connection-specific DNS Suffix . :
Tunnel adapter Teredo Tunneling Pseudo-Interface:
Connection-specific DNS Suffix . :
IPo6 Address.....................: 2001:0:9d38 :6abd:3434:1944:2684:c586
Link-local IPo6 Address ......... : fe80::3434:1944:2684:c586xl6
Default Gateway..................: ::

WesGilster, 2015-04-29:
So you are saying that the find printer host is working now that everything is on the same subnet?

kloknibor, 2015-04-30:
No it still isn't working... And the setup is exactly the same Like it was... The cmd opens and closes almost immidiatly... How can I provide more information for you?

WesGilster, 2015-04-30:
I need some details about what happened. If the cmd opened and closed quickly you should probably run this from a command prompt instead of double clicking on the link. That will give some debug information to work with. If it closed immediately, it either worked and you didn't notice your browser come up(that's happened to me a few times), it couldn't find the class or Java wasn't installed. If it couldn't find the class you should redownload the latest version. If Java isn't installed you need to install it.

Did you receive an error like this?
[image]

kloknibor, 2015-05-24:
It works now! bought a raspberry pi 2 today and testing everything again ;)! works like a charm now!

WesGilster, 2015-05-24:
Well, that's good news, you can close the issue when you are confident. I appreciate your testing! 

So do you think it's possible it's been working all along and you didn't notice what was going on? I've done this myself only to find my browser open with six different tabs browsing to the printer.

kloknibor, 2015-05-25:
How do I close issues :$? I only use github for this ;)

I'm not quite sure what was wrong... I got a new pi 2... And I recently reinstalled my whole computer... Installed every Java related thing (browser plug-ins jre sdk etc) and than it worked :D! I can't get my camera to work tough... And stl by link upload but Will look into that later/ maybe tonight

kloknibor, 2015-05-25:
closes #5

WesGilster, 2015-05-26:
I'll close.

closed 2015-05-26 by WesGilster