area515/Photonic3D #6: Display not being released

ergobot, 2015-01-21:
When successfully starting a printer, a display on the device will be used.  After stopping the printer and creating/starting a new printer on the previously used display, an exception is thrown stating that the previously used display is currently being used.

When successfully starting a printer, a display on the device will be used.  After stopping the printer, the display should be released.

WesGilster, 2015-01-30:
This is fixed in version cwh-0.62.zip.

kloknibor, 2015-02-17:
when stopping a job I get the following error : Error
Job:*filename*.cws not started

kloknibor, 2015-02-17:
after starting a newly created printer after stopping on the same display I get the following error : Display already assigned to:test(SerialPort:Console Testing, Display::0.0)

WesGilster, 2015-03-23:
I've attempted to recreate this bug with test console and display. I don't believe this bug still exists.

kloknibor, 2015-05-25:
Haven't encoutered this in a while... can be closes if you ask me!

closed 2015-05-26 by WesGilster