area515/Photonic3D #7: how to install :$

kloknibor, 2015-01-21:
Hi,

I've tried to install your software because I would be really intrested! but I can;t get it to work... I tried it with git clone... and I tried to use your networkinstall.sh script (by copying it into my ssh tool) but I can't manage to run start.sh on raspbian.... so what is the correct way to install? 

thanks in advance,

kloknibor

ergobot, 2015-01-21:
Have started assembling instructions:
https://github.com/area515/Creation-Workshop-Host/wiki

Will close this when instructions are more better for the pi.

kloknibor, 2015-01-21:
hi thanks for the quick support :D but unfortunately it still doesn't work... I used A fresh install of raspbian and followed all your instruction step by step... but the server doesn't come up... if you're on windows it's pretty simple to make a img of your succefull instal with win32diskimager you just give up a name and click read and you have a image to upload and make everyone happe (A) if you don't have time for that could you show me some screenshots? thanks for the great work :)!

kloknibor, 2015-01-25:
I managed to install iT... But run on startup doesn't work and I only get http 500 errors

ergobot, 2015-01-25:
You are really digging in, thanks for helping!
On startup, you will get that message.  The problem is being worked on.
The message means that there is no printer to communicate with.
You'll need to add a printer.  Enter a name for the printer, select a
display, select a serial port, and click the button create printer.
Similar to creation workshop, every printer has a display, serial port, and
a name.
Most of the stuff I'm working on is getting a burn and use image working
for the pi.  In case you want to give it a try in a non-pi environment, the
host works really well on a laptop too.

kloknibor, 2015-01-25:
I did all of that already and connected an marlin printer to it. The baudrate is 25000 so I'm guess that could be a problem? And when I load a .cws file and have uploaded the gcode it says it's running a job but the slices remain 0/0... That's as far as I got ;)

ergobot, 2015-01-25:
I'll check on the marlin and baud rate.  Are you using a muve printer?
Inside the .cws file, there should be the exported images for each layer
and one gcode (.gcode) file.  The .cws file is just a .zip file with the
extension renamed.  You can examine the contents of a .cws file with
anything that can open a zip file.
By default, CWS (creation workshop) doesn't export the slices.  On the
slice setting/printer configuration tab in creation workshop, you will want
to check the checkboxes for exporting slices as images.
As an alternate way, you can manually change the options to export images
in the profiles XML.  Off the top of my head, I think the two tags are
"export" and "exportSvg" in the XML.  Setting these two tags to "true" will
result in exporting the images when slicing.

kloknibor, 2015-01-25:
I'm using a home made reprap air (and a separate screen) at the moment only for testing purposes. I just funded the draken printer on kickstarter which is a dlp printer and on my old FDM printer I used octoprint which worked like a charm! but well it doesn't support dlp that why I'm so interested in this project! 2 features when everything works I really would like to see are access control (because I use to control my printer even when I'm not home so otherwise everyone could mess around with my printer) and ofcourse timelapse and video support ;)! but well thats for later on... lets first get this stable and I'm willing to help test as much as needed for now ;) The makers of the Draken dlp printer (3Dfacture) and the other backers of this kickstarter project are also really intrested in making their printer wireless! (I made some advertisement for your project already ;))

kloknibor, 2015-01-25:
and I will directly try your new suggestions ! I indeed didn't add the slices to the .cws file ;) you will hear from me :) how do I report errors the best way to you? just telling what I did en maybe posting screenshots? or how can I make it easier for you ;) log files somewhere maybe?

ergobot, 2015-01-25:
Thank you, any support is really appreciated as it helps make the project
better.  Github works great and we'll use it to report or comment issues,
features, bugs, and documentation.  Like the same way you reported the "how
to install" issue.
There are a few tasks that need to be done to make the pi image from
raspian.  I'll document those soon in case anyone can solve them.
Thanks!

kloknibor, 2015-01-25:
alright I will try to get the start on boot working and I will install the driver for the wn725n wifi stick ;) that's easier for me. Would it be useful to release a image from that with the software installed? than other people can test (easier) too.

closed 2015-02-01 by kloknibor