area515/Photonic3D #10: all displays are deleted

kloknibor, 2015-01-25:
hi,

today I installed the latest version of creation workshop host but with the new version (0.54) the displays are gone (so there aren't any displays to select in the list) so is it possible that somebody broke this by accident?

thanks in advance,

Robin

WesGilster, 2015-01-26:
I've been working on this quite a bit lately. Soon I'll have a complete 
start/install for the RasberryPi ready. Perhaps late tomorrow?

Thanks,
Wes G.

kloknibor, 2015-01-26:
would be nice ! thanks for all the hard work ;) and I will be ready to test ;)!

kloknibor, 2015-01-26:
and do you think you will make it complete today or is it save to go to bed ;P?

WesGilster, 2015-01-27:
Not sure if Ergobot has explained the development situation yet, but my 
repo is bleeding edge daily builds, area515 repo will eventually be 
stable builds. That means startdev.sh reads from my repo, start.sh reads 
from area515 repo.

If your raspberry pi is already capable of booting to a gui environment, 
then you are probably ready for this:
sudo wget 
https://github.com/WesGilster/Creation-Workshop-Host/raw/master/host/bin/startdev.sh
sudo chmod 777 startdev.sh
sudo ./startdev.sh

If your not booting your pi to a GUI and your depending on our print 
host to configure your X windows environment, you aren't ready for the 
above build(since I'm not done yet) and all of the displays will be 
missing and the print host will throw a bunch of http500 exceptions. I 
planned on getting this configured tonight, but not sure I'll have time.

Ergobot is looking into creating an image, but our start script will 
eventually take your Raspberry Pi from nothing works to everything 
working and upgrading by itself.

As you can probably tell this is changing daily, but here are the 
current instructions I think we'll be going with:
https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/installnotes.txt

Thanks,
Wes G.

kloknibor, 2015-01-27:
my raspberry pi starts in a GUI and in the log it says that the display variable is not set and sets it to 0,0 but it doesn't appear (anymore) in the webpage... one plus is that my printer gets recognized with your version :) but since I can't select a display it gives an internal server error

kloknibor, 2015-01-27:
I found out what the problem was... Normally I do everything by SSH but for some reason displays wont get recognized... but when I run everything from the raspberry Pi I can select the screens I need!

ergobot, 2015-01-27:
When I run through ssh, I have to export the display before running or in the command to run.

export DISPLAY=:0.0

ergobot, 2015-01-27:
After running 
export DISPLAY=:0.0

You will see a display named "0.0"

kloknibor, 2015-01-27:
ahh oke :) that step was missing ;)! by the way.... with printer simulation and console testing the printer is working at the moment! really excited :) it still can't control my marlin at 250000 baudrate but that doesn't matter ;) it isn't for this machine and I'll try to put sprinter at 125000  on it because the actual machine will run this and test farther ;)! the only thing else that doesn;t work for now is start on boot

WesGilster, 2015-01-27:
At the moment the only way to change the baudrate with the printer is to 
manually edit the file in the Machines directory for your printer. I'm 
working on the start on boot, I'm getting the service script copied into 
the /etc/init.d directory with the start script fine, but I haven't 
activated it yet. I'm still deciding on the most Linux portable way of 
doing that...

thanks,
Wes G.

WesGilster, 2015-01-28:
The new start.sh script now installs CWH as a Linux service(build 61), 
just make sure you delete the old /etc/init.d/cwhservice so that it can 
be installed properly on your next run. New users won't have to deal 
with the delete issue.

Now that this is done, it introduces more issues. For example, we are 
going to need proper logging (no more sysout), boot time is 
significantly longer if it decides it needs to install a new version of 
the app, and you won't have time to change your DISPLAY variable if the 
app starts on boot. I've put preliminary support for the DISPLAY 
variable, but I haven't had much success with it yet. Let me know how it 
works for you guys.

Thanks,
Wes G.

kloknibor, 2015-01-28:
I'm kinda busy tonight so I'll test iT tomorrow for you ;)!

kloknibor, 2015-01-29:
So I tested it on  a freshly flashed raspbian install ;) and well first time after running the script it worked great! but well... like you suspected after a reboot there isn't enough time to export the display... so the server can't seem to find display 0,0 so it will give an internal server error and just doesn't start any printer... but the automatic boot works ;)!

WesGilster, 2015-01-29:
Thanks for the test man this really helps us. Next step will be to add a 
service dependency upon xwindows. Could you send me the stack trace for 
that internal service error and the build.number text file you used. 
That HTTP500 should not have happened, instead, the displays should 
simply be missing with an error in the logs. I'll see what I can do 
about https/authentication this weekend, should be pretty easy to do. 
I'm also thinking about making an option to automate the lift speed 
settings a little bit. For example, what if I put together a linear 
function that determined the lift speed based upon lighted pixel 
coverage on the build plate? Something simple to start off with like:
y = mx + b
y = lift speed
m = pixel coverage for build platform(%)
x = maximum speed required to peel a full build plate - minimum speed 
required to peel a single pixel from the build plate
b = minimum speed required to peel any image

Obviously this would be an option, _not _a standard feature, because 
most people are placing these options into their creation workshop 
profiles and are gcode dependent...

thanks,
Wes G.

kloknibor, 2015-01-29:
It wasn't a http 500 error it just said "internal server error" nothing more nothing less, for Some reason this is only when the auto find printer function is used ;) about the trace stack... What is iT and where can I find it :p? Or doesn't iT help anymore if it isn't a http 500 error ;)! When I use a printer that's configured with display 0,0 than it just says the screen isn't available (only after reboot this occurs) so when we can get display 0,0 available for the server again than iT Will work again ;)! That's the biggest issue at the moment...

And authentication would be great :)! And your Peel technik sounds great too ;)!

If just one question left, I asked about the option to set the baudrate to 250000 and you said I could find iT in the machines folder, unfortunatly there's only the automatic printer in there... And not my own created printers... Could I use a own created printer for this purpose?

Thanks for the hard work! I'm also trying to work together with 3Dfacture so they can tell their backers when it works ;) this way your software will hopefully become popular :) and have a lot more testers or maybe developers ;)

Keep up the great work :)!

WesGilster, 2015-01-30:
I didn't fix the 500 error. "internal server error" is the definition of 
an http500 error. Sorry, I should have explained where the stack trace 
to the error will be found in the log.err and log.out files. Don't worry 
about extracting the log, you can just send me the files. When the 
xwindows dependency is fixed in the script, it will fix the display.

Sure why not? Make as many printers as you want and make each of their 
configurations the same, or different, that's what makes this software a 
bit distinct from others. You can create multiple AND start multiple AND 
control multiple AND watch multiple printers from a single host. However 
you obviously can't start two printers that share the same resources 
since that doesn't make any sense. So basically that means you can start 
as many printers as you have Displays and USB ports. That means the 
Raspberry Pi can only start 1 DLP printer since it has only 1 display. 
If it had 3 displays it would be able to start 2 DLP printers since it 
only has 2 USB ports. Starting printers are only constrained by 
hardware, you have no constraints on creating printers. So eventually 
when we start controlling FFF you won't require a display, just a usb 
port. You get the idea...

Thanks,
Wes G.

kloknibor, 2015-01-30:
I think I explained myself wrong, my printer is at a baudrate of 250000, last time I asked you pointed me to the machines folder to edit the baudrate manually ;)! So I went to the machines folder... But I only found the auto detect printer... Not the printers I created earlier... So I just want to change the baudrate of 1 of the created printers ;)! Not the automatic detect printer.

How long do you think iT Will take to fix the Xwindows dependencie? Because than I can sens the image to others to test ;)!

And where are the log files stored on the pi? 

Thanks

WesGilster, 2015-01-31:
They are stored in, the "Machines" directory under the user's home directory that is currently running the application. If you have created machines with different users, you can find them all like this:
cd /;sudo find -type d -name Machines

At the moment, log files are stored here: /opt/cwh

The changes themselves aren't that bad, but the key is to make changes that don't have further ramifications. For example, we are going down a road that forces everyone to start on boot and put the machine into a single user "Kiosk" mode. Many people, (me included) aren't interested in an application making those drastic changes to a linux install. Others like you, don't mind trusting us making all of the low level modifications to their OS and to make the Linux box only work with this application. It turns out we'll probably support both with different scripts, but requires some discussion to determine what we want to support.

kloknibor, 2015-01-31:
Whats's new in the update today with the r/d/ removed in start scrip? Does start on boot work now? (With displays)

WesGilster, 2015-01-31:
I'll be sure to post when it's ready.

Thanks,
Wes G.

WesGilster, 2015-01-31:
Display autodetection and start on reboot is working on RaspberryPi:2014-12-24-wheezy-raspbian.img.

This must be run on a clean image of Raspbian like the one suggested above, don't choose "boot to gui" in raspi-config and don't attempt to startx unless your up to the task of managing your own configuration.

Fixed in (build 65)

WesGilster, 2015-01-31:
I'm not sure how the image mucks this up, but without expanding your Root partition in raspi-config, X doesn't start right. WTF is with that?

I added a "Common Problems" section to the installNotes.txt here:
https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/installnotes.txt

kloknibor, 2015-02-17:
I guess this one can get closed :)

closed 2015-02-18 by WesGilster