area515/Photonic3D #9: error Couldn't find graphicsDevice called:undefined

kloknibor, 2015-01-25:
hi,

ergobot helped me a bit farther but I now still have the error that when booting up I get a 500 http error but this was known already.

when I create a new printer and select any display and click start printer I get the following error : 
Couldn't find graphicsDevice called:undefined

when I use autodetect printer and start a job I get the following error : 
Problem communicating with host printer. (http:404)

or : 

Printer not started:Autodetected Printer

and when I try to start the printer again I get the following error : 

Display already assigned to:Autodetected Printer(SerialPort:null, Display::0.0)

even after stopping the printer (when it's gives the same printer not started error)

My thought are it is related to the fact that I use a baudrate of 250000 and the other bug will have to with the other bug : https://github.com/area515/Creation-Workshop-Host/issues/6

hope this helps

AndySze, 2015-01-30:
I got the same error. SerialPort cannot be detected. Don't know why.

WesGilster, 2015-01-30:
I don't recognize the "SerialPort cannot be detected" error. Can you post the contents your log.err, log.out and build.number?

WesGilster, 2015-01-30:
I haven't been using the "Autodetect serial port".  Looks like that wasn't working. I believe the fix is in (Build 62)

WesGilster, 2015-02-07:
This autodetected serial port still wasn't working! It's fixed for good now. I've tested with the Raspberry Pi to be sure. Fixed in (build 79)

WesGilster, 2015-09-22:
I'm closing this bug and folding it into this feature:
3D Printer Firmware Detection #60

closed 2015-09-22 by WesGilster