area515/Photonic3D #12: Testing for Client Authentication

WesGilster, 2015-02-01:
config.properties now has a set of variables that control a simple user based authentication. I'll make the authentication mechanism pluggable once this is fully tested. Anyone what to test? (Build 67)

kloknibor, 2015-02-01:
yes! just tell me how :)!

WesGilster, 2015-02-01:
After you have build 67 installed, just change the following parameter in the config.properties:
useSSL=true

Then you can login with the username/password in that same file. I've only tested this in Windows...

kloknibor, 2015-02-01:
I use your repo and I'm rebooting as we speak so build 67 will get installed automatically right :)? adn where is the config file located? and do you want me any other bugs to test?

WesGilster, 2015-02-01:
Here is how it works. The latest stable version that is hosted on 
area515 will get installed automatically on everyone's machine on 
reboot. Unfortunately testers like you have to ignore that advice, 
because you have to test from an unstable repo to get the latest 
version(WesGilster). Once everyone feels like it's stable, I or Ergobot 
will promote it to the area515 repo. Until then you need to follow the 
"unstable" steps in the install notes.

At this point, I've modified all of the issues to show that they are all 
fixed and need testing. So yeah all of the issues on area515 need to be 
tested.

Thanks,
Wes G.

kloknibor, 2015-02-01:
does this mean that after installing with "sudo ./start.sh WesGilster" than if you reboot the software will get downgraded too the version from area515?

WesGilster, 2015-02-01:
Your actually on the right track, because that's exactly what it tries 
to do. However since the version on the area515 is always less than the 
version from WesGilster, that shouldn't ever happen. That is, unless 
someone corrupts the build.number again. If that happens, it basically 
wipes cwh off your disk completely.

Thanks,
Wes G.

kloknibor, 2015-02-01:
okay... and where is config.properties file saved?

kloknibor, 2015-02-01:
never mind found it ;) for the others it's in /opt/cwh stored along all the other cwh files :)!

kloknibor, 2015-02-01:
hmmm I just enabled the SSL functions and left all the other options default... this results in the server that's not booting... If I read the config file than it sounds like you need to buy a SSL certificate to use this function? or a CA certificat ... how to get one and how to install it ...? thanks!

WesGilster, 2015-02-01:
You can purchase a certificate, but you certainly don't need to. If it finds that a keystore isn't available and/or the cert is missing for the IP you are using, it will build everything for you. It will be a self signed cert and your browser won't like it the first time you go to the site. In Google you will need to "Proceed anyway" and your good to go with signing in with admin/password. If it's not booting, you'll need to send me all of the files that start with "log.". Are you testing with a Raspberry Pi? I haven't tried that yet. Are you sure it isn't just slow? Key creation is painfully slow on my fast PC so it's got to take an eternity on a Raspberry Pi. Eternity could mean minutes, I don't know, I've never tried.

Here is the bad news, if you have interrupted this process, you've probably corrupted the keystore and you'll need to delete it. You'll find the "web.keystore" in the /opt/cwh directory as well.

WesGilster, 2015-02-01:
Since it's a self signed cert, technically you should also check to ensure that the website cert fingerprint is the same as the cert in your keystore. Eventually I'll put the instructions up, if anyone is paranoid about this...

WesGilster, 2015-02-02:
Just tested this on the Raspberry Pi, and it seems to work great! Even the first time cert creation took much less time than I thought it would. I did perform another (build 68) because the config.properties had Windows line endings. Strangely, the build wasn't even necessary.

kloknibor are you sure you are hitting the right URL now? Did you remember to add the "s" in: https://[YourIp]:9091

If not, it will look like the server isn't even running.

If you don't know the URL to your Raspberry Pi feel free to start testing the "Automatic browse to print host (UPNP)" function. The idea is that you won't have to even know what an IP is to find your printer on the network. Of course it's not necessary for you, but it will be helpful for those that want to plug and play.

kloknibor, 2015-02-02:
Ah I'm sorry... I didn't use https indeed ... I've got a bookmark with the right IP and used that without thinking... Will give it another try ;)! And Will keep in mind the certificate creation Will take a long time ;)!

WesGilster, 2015-02-03:
I get your point; though, it'll work a bit differently in Jetty. Feel free to make an enhancement request, but I don't consider it a huge priority since a newb should simply double click on the browseprinter.bat/browseprinter.sh and it will open their system browser and their raspberry pi/host server is on the network. This will work even if the host is being given a different IP by a DHCP server, browseprinter will figure out where to find it.

Should I close this? I assume it's working for you?

kloknibor, 2015-02-03:
Yes iT works for me ;)! But I don't understand how your browsprinter.bat/browseprinter.sh works... How do I use that to find the pi in my browser?

WesGilster, 2015-02-03:
I'm not sure if you actually want to know how it works, or if you want to know how to run it? The directions should be in the install notes here if you want to know how to run it:
https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/installnotes.txt

If you are interested in knowing how it works, you can ask that question over on other issue "Testing automatic find printer host function (UPNP)" and I'll explain it over there.

closed 2015-02-03 by WesGilster