area515/Photonic3D #14: rpi slient boot - disable boot logging

ergobot, 2015-02-01:
Task: 
On the raspberry pi, disable all text when the raspberry pi boots.  Assuming the raspberry pi is using a new image of raspbian.

Background:
The goal is to start the raspberry pi (rpi) silently then start the creation workshop (cw) host.

A default install of raspbian on the rpi needs several modifications to accomplish this. This task is one, among many, to accomplish the above goal. After solving all of the tasks, the solutions will be put into a single script that will run once after a raspbian image is installed.

ergobot, 2015-02-01:
According to:
http://www.raspberrypi.org/forums/viewtopic.php?f=31&t=11642

Run the following command
sudo nano /boot/cmdline.txt

Replace "console=tty1" by "console=tty3" to redirect boot messages to the third console.
Add "loglevel=3" to disable non-critical kernel log messages.
To hide the text cursor, add "vt.global_cursor_default=0".

To see the console output anyway, just press Alt+F3 during or after boot.

closed 2015-02-01 by ergobot

kloknibor, 2016-06-17:
@jmkao This should get implemented in image generation. It will solve the need for a shutter and will help for the photocentric printer.