area515/Photonic3D #17: Raspberry Pi camera for build progress

WesGilster, 2015-02-02:
If anyone is interested in having a camera watch their build progress on their Raspberry Pi, buy this:
http://www.adafruit.com/product/1367
and ship it to me.

I'll see what I can do to integrate it into the host. I certainly am not going to guarantee video since I'm a little suspicious that we aren't going to have enough CPU to get this done, we might be able to take a picture every once in a while though...

ergobot, 2015-02-02:
I'm sure there's a way to do it.  RPI camera or web cam with images sent to
the browser (mjpeg)... something like that.
Wouldn't be worried about CPU (Moore wouldn't be):
http://www.theregister.co.uk/2015/02/02/raspberry_pi_model_2/

If anyone is interested in having a camera watch their build progress on
their Raspberry Pi, buy this:
http://www.adafruit.com/product/1367
and ship it to me.

I'll see what I can do to integrate it into the host. I certainly am not
going to guarantee video since I'm a little suspicious that we aren't going
to have enough CPU to get this done, we might be able to take a picture
every once in a while though...

kloknibor, 2015-02-02:
Take a look at octoprint, this software has already webcam integration on the PI :)! So it's possible ;)! This even works with cheap non raspbarry Pi camera's ;)! And sorry I need to save my money again... I'm just a student and I just bought a dlp printer... Altough I can test iT for you ;)! I've got a raspberry PI cam ;)! Maybe we can test it by teamviewer so you can do all the SSH work yourself ;) and extraxt the log files you need!

WesGilster, 2015-03-23:
Implemented in version .89

I should also mention this is completely untested. :)

WesGilster, 2015-03-25:
This isn't working. I forgot a key piece to make this work. I'll see if I can't get this working this week.

WesGilster, 2015-04-19:
Fixed now, start recording/stop recording of your build, then press the play button on the video. Pretty simple to use.

This will now support any raw h264 stream(many cameras out there) that you can send through to it. You will NOT need to install ffmpeg for this since everything is handled in Java native code. Everything is pretty much ready to go for the Raspberry Pi camera, the config.properties gives an example of how to use the pi camera at 100/100 pixels.

Recording in full HD worked like a dream on the new Raspberry Pi B+. If you are recording in HD and/or your recording long amounts of video you'll need to increase your max memory for Java. As you might expect, OutOfMemeoryExceptions are going to be much more common with video processing if you don't take precautions.

Working in version cwh-0.100.

If there is interest, I'll see if I can get real time streaming with DASH.

kloknibor, 2015-05-25:
I can;t get this to work... I use a new raspberry pi 2 with the original pi camera... When I try to record by  raspvid by SSH the red led start to light up so I guess it works... but not in creation workshop host... Altough I get an error about openbox (dont know what it is) ... could this be a problem for creation workshop host?

WesGilster, 2015-05-26:
I'll need logs to determine what happened. Where do you see the error about openbox.

SouthPaw-42, 2015-07-14:
CWSort won't display Pi Camera Picture but raspistill works from command line.

sbryan, 2015-07-14:
Odd, I just tested it last night and it was working for me?  All I did was install the camera on the RPi2B and booted up.  Then I click the "Record Video" button and I see the camera activate.  Later, I click "Stop Recording Video" and then i can click the "play" button on the embedded view and it replays what was recorded.

I did sort of think it was going to also stream the images to the web UI, but after re-reading the above, it seems that is not yet implemented?

SouthPaw-42, 2015-07-14:
Ok that record video works but the left frame still says "image not available"

sbryan, 2015-07-14:
The left frame will show you the current slice being projected, not any frame grabs from the camera.  Try uploading the calibration .cws file and do a dry run of the job (no resin, and don't forget the Draken has a print bed auto calibration sequence coded into the GCode before it does a print).  You should see each frame as it being projected also shown in that left image.

WesGilster, 2015-07-14:
It sounds like your camera is configured properly.  Can I get a bit more 
information from you on this? What steps can I take to reproduce this 
problem? In other words, which of the following steps doesn't work for you.

1. Press "Record Video" button
2. Wait for the desired amount of time.
3. Press "Stop Recording Video" button
4. Wait until the first frame of the recently recorded video is shown on 
the web page.
5. Press the "Play" triangle button on the newly created HTML55 video.

Thanks,
Wes G.

WesGilster, 2015-07-14:
Sorry apparently I'm responding to old posts...

SouthPaw-42, 2015-07-14:
@WesGilster Hi I thought you had some other features implemented. Would be great if you could take a picture with the Camera when specfic gcode is sent to printer. Example when the Build platform is raised and stop you could take a pic and put in folder then when print is done make a animated GIF from pictures. 
I would also display the most recent picture in the UI so you can chenck if the print has failed.

WesGilster, 2015-07-15:
What other features did you think I had implemented? Feel free to add another feature request! This sounds like a cool feature I'll mark this thread as a feature and rename the title. I'll integrate this feature with the resin level detection feature since it's really a subset of that issue. If you guys are interested in a stream feature, open another issue. I haven't worked on this in weeks because I didn't really see much interest or testing going on... Seems to be increasing a little bit now.

SouthPaw-42, 2015-07-15:
1.) Take a still picture every X seconds and display. Lets you check up on print progress.
X would be a field on the page.

2.) Take a sill picture every time a specfic gcode was send the picture would be displayed and stored. 
gcode would be a feild on the form.
When print is completed user would have option to make an animated GIF from pictures.

WesGilster, 2015-07-16:
I'll work on the taking still picture based on gcode this weekend along with the empty resin vat detection. Keep in mind, that we've really outgrown this gui and I'm really just throwing it together on the main page. The Gui rewrite will make things look pretty again.

WesGilster, 2015-10-03:
Backend functionality is complete, plus you can take pictures of different sizes now: cwh-0.136

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

New gui allows taking pictures while a print is in progress through the print job. Just click on the "Take Picture" button...

closed 2015-12-27 by WesGilster