area515/Photonic3D #19: error after signing in

kloknibor, 2015-02-04:
I guess it's the new build but I skipped some. I now get an error saying the following directly after signin in : Error Problem communicating with host printer. (http:401)

the printers,displays and ports don't show up and nothing works... what did go wrong? besides the self signed certificate gives errors in google chrome and it says the servers probably tries to steal data because the certificate couldn't be checked... but since its a self signed certificate there's nothing we can do about that, can we?

Robin

WesGilster, 2015-02-04:
Clear your cookies and try to sign in again. 401 is unauthorized which means you signed in with the incorrect username and password.I'll see if I can put a better error message on that.

There is something you can do. You can ask your browser to make an exception for this certificate or put it in the list of trusted certs. There is nothing wrong with this approach in fact the certificate is more secure than a chain of trust since there is only one private key that could be compromised(on your Raspberry Pi) instead of each private key in the chain of trust of CAs.However, you SHOULD NEVER indiscriminately accept certs before you compare the following two thumbprints:
On your host server in the cwh directory run:
"C:\Program Files\Java\jdk1.7.0_67\bin\keytool" -list -keystore web.keystore
In Chrome:
Click on the Padlock by the URL/Click Connection/Click Certification Information/Click Details/Click Thumbprint

If you are running this on a linux box, you'll have to find the keytool in some other location.
If you are using another browser, you'll have different instructions to find the thumbprint.

kloknibor, 2015-02-05:
maybe you can change it indeed into incorrect password and refresh after X seconds so they can attempt it again, I guess thats the most simple solution :)

WesGilster, 2015-02-05:
Fixed in (Build 72).

WesGilster, 2015-02-05:
From Kloknibor:
1. I enabled ssl and browsed to the printer
2. login screen came up and I filled in a wrong pas, the thin that happend was that the login screen just disappeard and came back blank (talking about milliseconds) so no error message
3. logged in with the correct password and got the following error "You logged in wrong. Please delete your cookies and try again." that's a nice error ;)
4. I deleted the cookies, as soon as I reached the printer page again it was logged in... So I didn;'t have to fill in any data at all to come in.
conclusion, it works.... but well its kinda... how to call it nicely... it just can be a bit better... but as concept it works :)

As much as I want to close this bug, I think we need to keep this open for a bit to see if I can reproduce this issue. If I can't, I may ask you to document this more specifically keeping close eye on the order of steps you performed. Thanks again for the testing. This helps quite a bit.

kloknibor, 2015-02-05:
Would iT help if I record my screen when I reproduce this error? Or need logs?

WesGilster, 2015-02-06:
Don't worry about it just yet. I'll try to reproduce it myself first.

Thanks,
Wes G.

WesGilster, 2015-02-07:
I cannot reproduce specifically what you are saying. Tested with both IE and Chrome. Both IE and Chrome both repeatedly continue to prompt the user for the correct password if it is put in wrong. This is normal behavior in all browsers. After three incorrect attempts IE will stop prompting the user for the username/password and show the "You logged in wrong message". Chrome, on the other hand continues to prompt for the username/password forever. In both browsers the cancel button cancels the login and shows the "You logged in wrong message". As to #4, this is completely normal behavior for Chrome, and is only normal behavior for IE if you haven't closed down IE completely. I have not been able to reproduce #3 at all. You say that the correct password causes the "You logged in wrong"? That doesn't make sense to me.

kloknibor, 2015-02-17:
I recorded today what I meant : http://youtu.be/i6AV8fYemME

WesGilster, 2015-03-23:
Need specific actions to reproduce this issue. Recording leaves out significant details. After finding that Chrome doesn't remove passwords for basic authentication, I'm skeptical that this is actually a bug.

closed 2016-06-21 by kloknibor