area515/Photonic3D #20: Donation by 3Dfacture for the hard work :)

kloknibor, 2015-02-05:
So I'm in contact with 3DFacture while they funded their new printer, they wanna thank you for the hard work and I can lower my pledge if I buy a camera and a raspberry pi 2 for you as a thanks to you guys ;)! Shall I just donate money somewhere or can I Send the components to one of you guys?

WesGilster, 2015-02-05:
That sounds great. I'm working on a websocket solution to eliminate the refresh buttons now, so I won't get to this right away. Do you know my email address already? If so, send me a quick message outside of github and we'll exchange shipping info privately there. I'll just have you order it and specify me as the shipping address.

kloknibor, 2015-02-05:
I don't have your email adress yet, contact me on robinkolk@msn.com so we can arrange it :)!

kloknibor, 2015-02-05:
Send you an email :)

WesGilster, 2015-02-05:
I've sent you a response.

WesGilster, 2015-02-06:
I believe we've settled this now.

closed 2015-02-06 by WesGilster