area515/Photonic3D #28: update over older version of WesGilster doesn't work

kloknibor, 2015-02-10:
So today I tried to update wesglisters build 69 to build 79 but when I try to reach the server the only thing it says is this : 

HTTP ERROR: 500

Problem accessing /index.html. Reason:

    Could not initialize class java.nio.file.FileSystems$DefaultFileSystemHolder
Powered by Jetty://

guess the updating turned out wrong, maybe because the old version wasn't closed down properly?
Will try again with a fresh raspbian install ;)

WesGilster, 2015-02-11:
It's hard to say what happened. HTTP500 essentially means that something bad happened inside of the server and you have to check the server logs. You do bring up a good use case though. Whenever I update, I always shutdown the running server first, delete my existing start.sh file and then run the three steps in the installation notes.

So this time, I didn't bring the old server down and started up another one. As expected, both versions started running with the new one in a crippled state that wouldn't take net connections. I suppose it would be a nice enhancement to perform a check to see if the old server is running and shut it down if it is.

WesGilster, 2015-09-21:
Implemented and untested in version 0.121.

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

closed 2015-12-27 by WesGilster