area515/Photonic3D #32: networkinstall.sh fails

rockjim, 2015-02-27:
The code references an old version and then fails because files aren't available.

Lines 7,8,12 should be "cwh-0.80.zip" not "cwh-0.54.zip"

kloknibor, 2015-02-27:
hi rockjim,

if you want to install it than take a look at this install guide : https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/installnotes.txt networkinstall isn't used anymore to install.

hope it helped,

Kloknibor

rockjim, 2015-02-27:
Thanks. I was going off of the
https://github.com/area515/Creation-Workshop-Host/wiki/Raspberry-Pi-Manual-Setup-Instructions
that I was direct to from the wiki.

kloknibor, 2015-02-27:
Sorry for the outdated wiki ;)! I'll try to update it this weekend :)

WesGilster, 2015-03-23:
Documentation really needs cleaned up on the Wiki. So old. Maybe some screen shots to peak some interest, now that the camera, current slice and total cost are implemented.

kloknibor, 2015-05-25:
Copied the description from the homepage to the wiki for now... will try to make a install clip when I have time

WesGilster, 2016-06-23:
Since this bug was opened I think you've made two different install clips. Feel free to reopen if you don't want to track your progress under #208.

closed 2016-06-23 by WesGilster