area515/Photonic3D #33: Camera progress

kloknibor, 2015-03-11:
Hi,

I haven't tested much lately because I'm kinda busy with my study at the moment, but already started with testing the webcam :)? and recieved the new pi ? does this make the slicer you build a bit more usefull?

I hope I can start testing next week ;) if there's anything particullary to test name it and I will test it ;)!

Robin

WesGilster, 2015-03-12:
My slicer and webcam will probably be ready to test at the same time. At 
that time you can test both. In the mean time, the following two things 
would be great:
1. Network interface(ip address/subnet) information for your raspberry 
pi and the client you are using to connect to it from the "autodetect" 
client.
2. Step by step instructions for recreating the password login issue. 
Don't worry about your English, it's just fine. Focus on simple 
instructions from start to finish. I can recreate the problem by a 
series of installs and password changes, but technically it isn't a bug. 
It turns out that Chrome doesn't clear username/password info for sites 
utilizing basic authentication, no matter what options you check when 
you clear history. (A serious security concern if you ask me.) You can 
verify this for yourself by checking to see if the following is a header 
in your requests after performing a password clear:

|Authorization||  Basic ZWx1c3VhcmlvOnlsYWNsYXZl|

Thanks,
Wes G.

WesGilster, 2015-03-23:
Closing since it's a duplicate of another bug.

closed 2015-03-23 by WesGilster