area515/Photonic3D #34: Graph function simulation for all variable calculators

WesGilster, 2015-03-25:
Now that you can create your own functions. We need to be able to graph those function with the touch of a button so users can see how their functions break down over variable ranges. This should operate at the touch of a button.

WesGilster, 2016-01-03:
Fiixed in cwh-0.183.

jmkao, 2016-01-17:
These work pretty well, I haven't found any issues with the graphing, although it took me a little while to figure out that the layers chosen for the two lines are at the boundary before and after the initial layers.

I am in the process of writing a wiki page describing how to configure CWH for a new printer that describes the use of functions and the function simulation:

https://github.com/area515/Creation-Workshop-Host/wiki/Slicing-With-Creation-Workshop-Host

WesGilster, 2016-01-17:
Whoa. Holy cow your getting serious now man...

WesGilster, 2016-01-17:
You mentioned that the two lines were specified at the boundary, but the GUI can be expanded in a ton of ways by specifying different variables. In fact, it's nothing more than adding some text boxes and filling in the proper Angular expression.

You simply just need to call the js function "testScript". It takes two parameters:
1. The name of chart window that pops up after the test is complete.
2. The formula for the return values that get fed into the chart.
The formula takes the form:
returnType
or
returnType[$chartDomainVariableName(startingValue, stoppingValue, steps),$chartRangeVariableName(startingValue,stoppingValue,stepIncrement)]

In CWH, I've hardcoded the the Angular expression for the Z Lift Distance Calculator to look like:
printersController.testScript('Z Lift Distance Calculator' , 'java.lang.Double[$CURSLICE({{printersController.currentPrinter.configuration.slicingProfile.InkConfig[printersController.currentPrinter.configuration.slicingProfile.selectedInkConfigIndex].NumberofBottomLayers}},{{printersController.currentPrinter.configuration.slicingProfile.InkConfig[printersController.currentPrinter.configuration.slicingProfile.selectedInkConfigIndex].NumberofBottomLayers+2}},1)$buildAreaMM(0.0,2000.0,75)]', printersController.currentPrinter.configuration.slicingProfile.ZLiftDistanceCalculator)"

kloknibor, 2016-03-07:
This works fine, only the gradient bulb thing shows a bunch of code and that it works... Maybe an image of the mask might be nicer?

nerginer, 2016-03-26:
Is there a way to use the bulb mask as image file? If so where will I put that file and how to define in CWH to use that image for bulb mask?

WesGilster, 2016-03-26:
No there really isn't. Let me make you a deal. Take a look at the process that jmkao has put together here:#109 and 1 of 3 things will happen.
1. You will hate it.
2. You will think it's ok, but realize that it needs a graphical interface tied into CWH itself to eliminate the equation editing.
3. You will love it, because you will realize that your projector already supports bulb hours querying and you could apply that coefficient directly to your equation and not have to worry about generating another png mask when you replace the bulb or find that your current mask isn't measuring up due to this: https://www.christiedigital.com/TechPapers/Christie-Lamp-Life-Explained.pdf

If 1 happens I'll add the png mask this weekend.
If 2 happens I'll add prioritize integrating the bulb mask creator into the GUI.

nerginer, 2016-03-26:
Thanks Wes,
how do you know that you design the perfect mask with projector mask creator? I am able to read the **uv exposure** in any place on the vat with my cheap uv sensor. Can we make this feature better by getting values with this sensor?

[image]

WesGilster, 2016-03-26:
Yeah. With that it would be pretty easy. Essentially you need to take two measurements. Highest and lowest. 
Lowest measurement will be in some corner: L
Highest measurement will be in the center of the bulb: H
Lowest measurement will be based upon the your bulb hours coefficient: C
Highest measurement will be: (H - L) * C

WesGilster, 2016-03-27:
Sorry wrong again...
Highest measurement will be: (H/L) * C

nerginer, 2016-03-28:
Great! The only thing I am not sure is how do you determine the exact position of the bulbCenter and bulbFocus.   In projector mask creator we just click with our mouse.

Sorry one more; Where do we define the bulb hour coefficient Does CWH query the projector and got it automatically?

kloknibor, 2016-03-28:
I only know the answer to your second question : Yes CWH gets the bulb hours automatically from the projector ;)!

WesGilster, 2016-03-28:
You know the exact position of the bulbCenter because that will be the location that you find highest possible UV measurement. I'll make a quick image mask creator inside of Photonic3D where you'll be able to move an 'X' in the locations where you are taking measurements with your UV sensor.