area515/Photonic3D #35: Show 3d image for STL with non-watertight regions highlighted in red

WesGilster, 2015-03-25:
The functionality for this has already been developed server side. We just need the following client side changes.
1. Ask for the STL file from the server
2. Asychronously ask the server to perform a slicing simulation and stream the triangles back for all non-watertight regions as the simulation finds them through a websocket.

WesGilster, 2015-05-23:
I'm really pretty disappointed in OpenJSCAD. It just sits there and grinds on large stls. I've built out the server side and it's ready for everything now, including a simulated print where each layer reports back the errors, but I'm having some trouble finding a good Javascript tool to show the model. I've thought about shooting back all of the triangles through a websocket and displaying them as they flow in from the web socket, but then it will take even longer to show the full model than the "all at once" idea that OpenJSCAD is built for.

I suppose I could shoot back only the triangles that are involved in the z slice, but I'm not sure that would be enough of a model to give the user enough perspective to analyze the highlighted regions.

WesGilster, 2015-05-23:
Any suggestions?

WesGilster, 2016-06-11:
I've been bragging about this one long enough, I finally put this guy together. Basically it's a custom Angular directive built to show printJobs geometry. To test this out, I've built a nonwater tight cube with lots of "holes" in the geometry, and checked it into the project. I've also added a feature to pause the print when bad geometry is found and highlight the triangles involved in the poor geometry. What I did was build a quick simulated printer configuration with no exposure time. Then the simulated print zips right up to the first non-water tight region it finds:
[image]

kloknibor, 2016-06-12:
I just tested it and it works nice! Great addition Wes thanks! Altough I'm thinkin it might be nice to add this function as previeuw option of the uploaded files too. If I got time later today I will push my proposal ;)!

kloknibor, 2016-06-12:
Hmm what I was trying to do was to add an previeuw button in printables.html Adding the button is no problem ofcourse. But I'm not quite familair with angular and couldn't find my answer in their documentation yet. 

What I tried is as to do an ng-include on 3dmodel.html but it remains hidden. I was considering 2 ways of implementing. 

First was to add an 3d view card to the right of the printables. 

Second was to do an pop-up like upload.html so working with ng-click. But to be honest I didn;t understand how I could add the function to printables controller. Could you explaing me?

WesGilster, 2016-06-12:
Sure that would be awesome, but you need to perform a little design work first. Take a quick look at 3dModel.html and you will find it's based on print jobs and not printables as shown here:
<print-job-model-viewer print-job="printJobsController.currentPrintJob"></print-job-model-viewer>

This is important because if you don't have a print job, you won't be able to call the geometry functions built into the PrintJobService. Easy fix of course, because that information can be streamed directly with a very standard STLLoader built into THREE.js. In fact it's significantly easier than the custom geometry loader I built. 

PrintJob is a nice object because it self contains all the information necessary to show common geometry information, printer buildarea information, and current slice information. On the other hand, a Printable object(by itself) has none of these things. For example, with printables, geometry information is file specific, doesn't have buildarea information and doesn't have slice information.

To summarize.
Step 1: You'll need to fix up the printJobModelViewer directive a bit so that it can take either:
1. A stand alone PrintJob (Already done)
  or
2. A Printable and Printer.

Step 2: You'll need to fix up the 3dmodel.html to show a printer dropdown list(preferably in the header) so that it's able to show build area information which I'm working on next.

Step 3: You'll need to fix up the printJobModelViewer directive to allow it to take more than just my custom json file type. I'll build a restful file download that you can use for this.

To answer your question, don't worry about ng-click it isn't the right direction for Angular and that makes things complicated. With Angular you "watch" variables, and when they change, your directive will change with them. That is the double binding you hear so much about in Angular. Instead you'll write something like this:
<print-job-model-viewer printable="printablesController.currentPrintable" printer="selectedPrinterFromDropdown"></print-job-model-viewer>

That's it, no functions needed. Then when the currentPrintable is changed in the gui whenever and for whatever reason, your 3d model will change automatically.

WesGilster, 2016-06-23:
Since the goal of this bug is complete, I'm closing. I realize it may still have good information that we can refur to from #133, but that conversation should probably continue in that thread instead of this one.

closed 2016-06-23 by WesGilster