area515/Photonic3D #36: Server pull STL file from Internet instead of performing upload

WesGilster, 2015-03-25:
Easy change, but I think it would be really cool to pull the image from a URL rather than forcing the user to download the image to their local harddrive and then uploading the image to CWH.

WesGilster, 2015-05-03:
This works great now. Also fixed a classpath problem and binary stls should work as well. You'll be notified through a websocket message that the JobStatus is "Ready" to be printed.

Try it out by uploading this file:
http://www.thingiverse.com/download:888699

The functionality is available in cwh0.108

kloknibor, 2015-05-25:
I get an http 500 error and that the printer is unresponsive... (I haven't connected a printer so far) how can I provide you with more debug information?

WesGilster, 2015-05-26:
Again, I'll need the logs for this to know more.

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

closed 2015-12-27 by WesGilster