area515/Photonic3D #37: Skip build number check if GitHub is not responsive

rockjim, 2015-03-29:
Today with GitHub under DDoS attack, I'm getting stuck when starting up because the start script is trying to check for a newer build.  The request keeps timing out and only occasionally connects.  The problem of GitHub begin down may be unusual, but it seems there should either be a two try limit to connect, or an option to override when starting the service.

WesGilster, 2015-03-29:
This shouldn't cause the script to fail. Does the whole start script freeze then? What is the nature of the issue? What do you mean getting stuck? It would be pretty easy to change the timeout from 90 seconds to something shorter if this is a performance issue.

http://www.gnu.org/software/wget/manual/wget.html
--timeout=seconds

rockjim, 2015-03-30:
Sorry for not being more clear.  When I was trying to run start.sh it would
try to wget the build.number and then timeout because of the problems
github was facing.  Once it timed out, it would then do attempt #2, then
#3, etc.  My point was that especially after 2 attempts failing, I think
the script should stop attempting to connect to github and just run the
currently installed version.

If there is a way there was an command line option to bypass the build
number check when running start.sh, I missed it.

WesGilster, 2015-04-06:
Fixed in cwh-0.97. Timeout is now set to 20 seconds with two retry attempts. Let me know how it works...

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

closed 2015-12-27 by WesGilster