area515/Photonic3D #39: Gcode file was not found in print directory

mcguinnessdr, 2015-05-22:
When I try to print a cws file I get the error: "Gcode file was not found in print directory". Anyone know what might be causing this?

ergobot, 2015-05-22:
Sure, what steps you took leading up to it and will take a look.

mcguinnessdr, 2015-05-22:
I simply uploaded a cws file, started my printer, and hit "start job". I could possibly be doing something wrong, as I just go my pi yesterday, so don't really know what I'm doing.

kloknibor, 2015-05-22:
Hi,

I made a clip of how to use CWH maybe this helps you ;)!

http://youtu.be/J3HTCkxlKcw

If it still doesn't work let me know!

mcguinnessdr, 2015-05-22:
Thank! I believe the problem was caused by me not exporting the slices to the cws file. I will have to check if it works after I get off work though. Were there instructions somewhere that said this was necessary that I just missed?

closed 2015-05-22 by mcguinnessdr

WesGilster, 2015-05-23:
I wanted to give a hint to the user that they haven't exported their scene properly. Now the error will be:
Gcode file was not found. Did you include the Gcode when you exported your scene?