area515/Photonic3D #40: Problem communicating with host printer. (http:500)

eccce, 2015-06-13:
Hi. I have problem with connecting CW Host to my Ramps 1.4.    I get massage "Problem communicating with host printer. (http:500)" the moment enter the Host.
[image]
   I am using Marlin firmware with baudrate 115200.

WesGilster, 2015-06-14:
> screenshot 2015-06-13 14 13 15

You'll need to send the following two files for me to determine your 
problem:
log.out
log.err

Thanks,
Wes G.

eccce, 2015-06-15:
Please see files on Dropbox 

https://www.dropbox.com/sh/x3nuqrft059o6uy/AACLvcS2dGOKCKF5nu-CcosNa?dl=0

WesGilster, 2015-06-16:
You must have restarted CWH after the error happened and cleared the 
logs. Every time that you restart the CWH the application/error logs are 
cleared automatically. This means that you have to recreate the 
problem(HTTP 500) and then send the logs.

Thanks,
Wes G.

eccce, 2015-06-16:
In which directory should be those files? I use search but it is very slow...

eccce, 2015-06-17:
Please see updated files on Dropbox link posted above.

Thank you.

WesGilster, 2015-06-20:
I found this in your logs:
org.area515.resinprinter.display.InappropriateDeviceException: It 
doesn't look like your graphics environment is properly setup
Can't connect to X11 window server using ':0.0' as the value of the 
DISPLAY variable.

This most likely means that you didn't follow the instructions in our 
setup, or you used an existing Raspberry Pi install where you've already 
tinkered with the display settings. Maybe at some point in time you went 
into raspi-config and chose "Enable Boot to Desktop/Scratch". That 
option pretty much breaks CWH due to how it wants to use the display.

Here are your options:
1. If you are a new to Linux and don't really want to fuss with setup 
options and you just want CWH to work you'll need to reinstall Raspbian 
from scratch with these instructions:
https://github.com/area515/Creation-Workshop-Host/blob/master/README.md
2. If you are a strong Linux admin then you've already seen the problem 
above and know how your x11 server is configured and how you'd like to 
integrate CWH into your custom XServer setup. Keep in mind, that one 
gotcha is that the setup.sh script will be overwritten on every new 
version released of CWH.
3. If you've only done some basic Linux admin work and you absolutely 
must boot up to a gui, then you'll quickly find that installing CWH as a 
service(which is what we do) might not be what you are interested in. At 
this point you'll need to ask some questions:
     a. What is the main function of my Raspberry Pi?
     b. How and when do I want CWH to start?
     c. What user is CWH going to run under now?
Next you'll probably need to learn about how to do these things under 
whatever Window Manager that you happen to be using.

Thanks,
Wes G.

WesGilster, 2015-07-07:
The 500 error is fixed in version cwh-0.109.zip. This will not fix your underlying pi setup problem, but you won't get the 500 error anymore.

WesGilster, 2015-09-22:
No further comments on this issue, I assume that the existing instructions are sufficient.

closed 2015-09-22 by WesGilster