area515/Photonic3D #41: [Question] Will CWH development continue?

sbryan, 2015-07-02:
Now that CW is no longer opensource [1] (Steve has pulled the repo and has indicated he will no longer be providing sources), what does the future hold for CWH?  I would hope it can stand on it's own, but I wanted to hear it from those still working on it.

[1] http://www.buildyourownsla.com/forum/viewtopic.php?f=5&t=3066#p10226

ergobot, 2015-07-02:
Creation-Workshop-Host (CWH) will remain on github.  CWH is a seperate
application.  CWH isn't going anywhere, we'll keep working at it.

Now that CW is no longer opensource 1
<http://Steve%20has%20pulled%20the%20repo%20and%20has%20indicated%20he%20will%20no%20longer%20be%20providing%20sources>,
what does the future hold for CWH? I would hope it can stand on it's own,
but I wanted to hear it from those still working on it.

[1] http://www.buildyourownsla.com/forum/viewtopic.php?f=5&t=3066#p10226

sbryan, 2015-07-03:
Thanks, great to hear... I'll be trying to get this to work soon on my just delivered Draken printer (from the 3D Facture folks that knibor was helping to enable this for wireless printing.  Closing the issue...

closed 2015-07-03 by sbryan