area515/Photonic3D #42: Determine when resin in vat is empty

WesGilster, 2015-07-07:
I ran out of resin on a print that took 6 hours. That's pretty annoying. Integrating the raspberry pi's camera was annoying with video, but was simple with the still camera. Now I'm thinking about putting the still camera to practical use. I'd like to detect the edge of the resin and determine when the resin drops below a certain level, we pause the print, or send some gcode to fill the vat. I'm thinking about placing a transparent sticker with a red circle and determining when the liquid level falls below that circle, we are out of resin. Seems pretty straightforward. I'm thinking the code should be able to plugin from another project and just work. Any interest in this?

WesGilster, 2015-07-19:
The test for this is working unbelievably well. I took a picture of a semi transparent alcohol bottle with a sloppy handdrawn circle on the side. The camera was upside down, a small tilt, and wasn't aligned perfectly with the bottle and the lighting quality was good. The liquid level detected percentage was better than what I could determine on my own.  I could simply see it was just barely over 50% and the visual processor determined it was 56%. Of course without further tests, it's hard to tell if it's a coincidence.

The visual processor implementation is a sloppy generalization of the Hough's transform which started from:
http://rsb.info.nih.gov/ij/plugins/download/Hough_Circles.java
It doesn't really look much like the original class and the threshold detection actually works as opposed to the original which was confusing and had to be calibrated with different theta samples.

The edge detection is a Grayscale implementation of the CannyEdgeDetector by Tom Gibara.

It certainly doesn't perform fast, but there are a ton of optimizations you could put into it.

sbryan, 2015-07-20:
Just curious if you looked into using OpenCV for any of this?  It has implementations of many of the algorithms you mentioned above and there is a [Java version of the OpenCV library](http://opencv.org/opencv-java-api.html) available.

As quick set of pointers, here are links to some of the APIs that provide the algorithms you mentioned you used and/or would likely need to use:
* [Convert color](http://docs.opencv.org/modules/imgproc/doc/miscellaneous_transformations.html#cvtcolor) spaces, for example, to grayscale as needed for the HoughCircles() API below.
* [Gaussian blur](http://docs.opencv.org/modules/imgproc/doc/filtering.html#gaussianblur) to lower the incidence of false positives in the circle detection.
* [Hough Circle](http://docs.opencv.org/modules/imgproc/doc/feature_detection.html#houghcircles) detection from gray scale image.
* [Canny](http://docs.opencv.org/modules/imgproc/doc/feature_detection.html#canny) edge detection
* [Hough Line](http://docs.opencv.org/modules/imgproc/doc/feature_detection.html#houghlines) detection, which, when applied to the output of the above Canny() function, will produce a vector if line segments.

Also, as I understand it, If the target platform has OpenCL optimizations available, they can be utilized as well to boost performance.

Just an idea to look at, as it may lessen the amount of time/energy you need to apply to support and maintain what you have written based on the same algorithms.

WesGilster, 2015-07-20:
Yeah, I did, OpenCV is pretty top notch and covers quite a bit in the 
world of computer vision. I'll be honest, I really hate managing native 
code libraries. With this project, all I have is Serial and Java itself, 
and it's been a pretty big pain. I know the promises are always great, 
depend on the OS packages and install them... So how do I do that with 
Oracle Java? How do I do that on Kali Linux(which I use at home on a 
Raspberry Pi) with RXTX?  Other projects I've worked on have the same 
issues, Labjack, Arduino. Pain pain and more pain... Maintaining 12 
extra classes are nothing compared to the dependency agony that ensues. 
However, if someone else would like to maintain the native CV 
dependencies and how to install them on each OS, I'll be the first to 
use them. :)

Thanks,
Wes G.

sbryan, 2015-07-20:
Makes sense... Again, I'm exposing my lack of familiarity with JAVA based projects, how they build and how they are distributed.

I can find installable packages for Ubuntu (libopencv2.4*) and there are [upstream debian packages](https://packages.debian.org/search?keywords=libopencv) as well.  Since Kali is debian based, should be pretty easy to figure out the deps and add an `apt-get install $OPENCV_PKGS` where OPENCV_PKGS is set to what ever the deps list is.  Then in the start.sh, where a test for librxtx-java is made, do something similar for opencv...

Again, I'm probably way over simplifying the issues, and was by no means trying to suggest a "better" way.  More over, since I can't really offer any help in the build/package parts, I'll just accept what those who do manage that decide ;)

I have done some package creation and management in the past for other linux distros that used RPM style packaging, but it's been some time.  Sadly, I'm not well versed on debian packaging concepts, otherwise I'd offer some help in that regard so you could have a cwh.deb installable package that had it's install and run dependancies resolvable and installable during package install time.

jmkao, 2015-07-20:
I don't think it's an issue with Java based projects, more the difficulty with cross-compilation and portability of native libraries. I've poked at the CWH code on x86 Linux and Windows machines, and if I modify items, I can simply copy the installation binaries over the the RPi and run them  using the ARM Java there.

Once there are native libraries, then the question of matching environments between development and deployment environments comes in, and if you want to get an exact matching set of library versions, you need to enter the dark world of cross-compilation (to build binaries on a more power x86 Linux box) or have to start doing the compiling on the RPi itself, which is slow and resource limited.

Some of this can be mitigated by having the flavor of Linux you run on the RPi being similar to the flavor of Linux you develop on (e.g. Debian on both, where packages are similar between different architectures), but then you have the case where you have to maintain not just OS but distro parity across any box you might be developing on. This is party why OctoPrint is far easier to install as a whole RPi image (where you get the whole OS environment in addition to the app) compared to trying to install it as a package with all its dependencies.

WesGilster, 2015-07-20:
jmkao nailed it. Of course it can be done, it's done every day and I use those packages every day. On the other hand, those packages are also brittle, broken and outdated every day(try an apt-get on rxtx on Kali). When that happens I have to become a package maintainer, cross-compiler and Unix admin just to get my work done. Then after I've done all of that, I have to figure out how I integrate that in the autoupdate script we have running. I'll take maintaining 12 classes over those tasks any day, especially when only 2 or 3 of those computer vision classes are intense. If the only factor was time and I was getting paid to make these decisions, I might make another choice, but computer vision maintenance will always be more fun than deployment decisions. :) 

As a side note, due to being pure Java, I took this a bit more seriously:
http://imagej.nih.gov/ij/

Notice that is where I got the inspiration for the generic hough implementation I mentioned above.

MongerDesigns, 2015-08-07:
Hi Wes,

Love what you've done with this project. We need some custom work done with this, mostly modifying the UI and some of the buttons. Please contact me at mongerdesigns@live.com
Thanks.

WesGilster, 2015-09-24:
Finally got some time to build a test kit build and added the ToughSituation test as a part of the suite.

Here are the results from the latest Raspberry Pi:
Time taken to perform visual inspection of remaining print resin: 5599ms
Remaining print resin: 56.36%

WesGilster, 2015-10-03:
This is complete with tests changing tags.

WesGilster, 2016-06-23:
This is implemented and put into production long ago.

closed 2016-06-23 by WesGilster