area515/Photonic3D #47: Add usage of raspi-config-cwhost.sh to installation documentation

sbryan, 2015-07-07:
I'm not sure what the state of this script is, but having just yesterday run through the instructions in README.md and wiki/Raspberry-Pi-Manual-Setup-Instructions, I knew nothing of this script until looking for something else in the repo this morning.

I note this because my experience of installing yesterday did not seem to match all the notes... the Raspian boots to the graphical desktop anyway, despite having selected not to in raspi-config... yet, based on the notes in README.md, I expected to be at a non-X console only interface after bootup.

Anyway... having read through this script (`raspi-config-cwhost.sh`), it looks like it would solve several of the inconsistencies I've noticed, so when I am home later, I will try it out and see how it works.

Just thought that if you all think it *IS* working, then maybe it ought to be added to the README.md and wiki pages so folks don't miss it like I did ;)

WesGilster, 2015-07-08:
So really, there are two situations here:
1. What are the minimal alterations that it will take to make CWH operate under 'most' Linux environments.
2. What will it take to take make your Raspberry Pi a dedicated printing board and disable all of the stock features that come with your the Raspberry Pi window manager.

start.sh 'attempts' to perform task #1 
raspi-config-cwhost.sh 'attempts' to perform task #2 (However it 'should' be optional)

If you've found something that breaks either of these two conditions I'd really like to know about it, and so would the community.

sbryan, 2015-07-08:
Not a "break" per-se... I was simply confused since the README.md and wiki page do seem to start off specific to Raspberry/Rasperian usages... seems like even a footnote along the lines of "If you are setting up a dedicated wireless print server for your 3D printer, you can also run the `raspi-config-cwhost.sh` script to do so... warning, blah, blah, blah..."

If the intention is to protect users from accidentally doing this to their Pi, then, well, I guess this is not a bug ;)  I'm only suggesting that it you wanted folks to know about the script, it would probably help to document it someplace :P   Otherwise, feel free to close as "not a bug", I won't be offended.

WesGilster, 2015-07-08:
Yep, I'll document that. It's just that I'm not sure that script is heavily tested. Please try it and let us know what you think.

Also, please let us know if the minimal procedure doesn't work for you. We need testing more than anything. If you haven't noticed, we are lacking in that department...

WesGilster, 2015-07-08:
I'll also move the raspi-config-cwhost.sh in the bin directory so that it is more accessible...

WesGilster, 2015-07-08:
I organized the information a little bit. I mentioned and provided a link to the raspi-config-cwhost.sh script as well. Added a few bits of information that might be helpful as well.

sbryan, 2015-07-09:
Well, I messed around with this for several hours last night, and, to be honest, I think it messed things up more than helped... nothing personal.

It may be that enough has changed between the 2015-02-16 and the 2015-05-05 raspian images that the script needs to do more context checking.  All I know is that I installed raspian from the NOOBS image that contains a newer version than what your instructions list... otherwise I did everything by the "book"

What I found was that it was calling `mkdir -p` in several places where the path contained the actual file that was going to be created, not the directory it was contained in like it should have been... here's an example:

    do_hide_mouse() {
      sudo apt-get install unclutter
      mkdir -p /home/pi/.config/lxsession/LXDE/autostart
      echo "unclutter -idle 0" >> /home/pi/.config/lxsession/LXDE/autostart
    }

In the above function, the result is a *DIRECTORY* named `autostart`, which is not what you want, as the following `echo` command throws an error:

    ./raspi-config-cwhost.sh: cannot create /home/pi/.config/lxsession/LXDE/autostart: Is a directory

This also happens in `do_autostart_cwhost()`

In addition to the above, the script requires that it be run as root (sudo), yet it is explicitly modifying the `pi` users configs with root permissions on files, but, at least on my installed image, the default Xsession is running under root, not pi?

Again, much of this could also just be some session/desktop config changes in the later version of raspian I am using.

I am burning a clean image of the 2015-05-05 raspian to my microSD card now.  If you want me to debug any of this on a clean install, I can try, just let me know what info you may need.

WesGilster, 2015-07-09:
No need to apologize. I didn't write it and have never run it. I'm not 
sure it has ever been executed...

Thanks,
Wes G.

jmkao, 2016-03-11:
I think this is largely superseded by the RPi image generation and the way start.sh works. I'm closing this, free free to re-open if you feel that I got it wrong.

closed 2016-03-11 by jmkao