area515/Photonic3D #48: Missing license and copyright information

sbryan, 2015-07-07:
I'd like to contribute, but, as SW developer by trade, my company requires me to disclose to them any plans to contribute to open source projects as a standard practice of their IP protection.  As part of that process, I need to provide any known license and/or copyright information about the projects to to which I'd like to contribute.  I can't seen to find any licensing or copyright info in the repo...

Also, I'm assuming the answer is no, but do you require any kind of signed contributor agreement like some projects do (Android, Cordova, WebKit, etc...)?

WesGilster, 2015-07-08:
1. There isn't any signed contributor agreement.
2. We don't have a license, but I've been wanting to adopt one. If you have a good copyleft license that is compatible with your company, feel free to suggest it. It's a little late now, but better late than never.

sbryan, 2015-07-09:
IANAL, so, I can't make any recommendations that are really worth anything than what you paid for them :p 

If it's copyleft you want, I suppose I'd say CC-SA.  Just remember, IANAL, so I've really no way to know what the relative merits or risks are between them.

All I can say is that many Corporate types don't like some of the GPL versions (don't recall exactly which one(s)) because they consider them "viral"...

Just make the best decision for what you think you want, and go from there... the good thing is, as long as you are the original author(s), you can change the license whenever/however you want ;)

WesGilster, 2015-07-14:
Hey Sean O'brian(Ergobot), what do you think of this... Allows derivative works if they are shared, and allows commercial use.

[image]<br /><span xmlns:dct="http://purl.org/dc/terms/" property="dct:title">Creation Workshop Host</span> by <span xmlns:cc="http://creativecommons.org/ns#" property="cc:attributionName">Wes Gilster & Sean O'Brian</span> is licensed under a <a rel="license" href="http://creativecommons.org/licenses/by-sa/4.0/">Creative Commons Attribution-ShareAlike 4.0 International License</a>.

sbryan, 2015-07-15:
Once you guys sort out the license, I'll be able to send PRs, assuming you stick with a copyleft-ish or BSD style license.

You might also want to add copyright headers to all your own source files at least, and, depending on the licenses, to those you've imported and modified... Ain't lawyers great! ;) :-1:

ergobot, 2015-07-15:
License added.  Creative Commons Attribution-ShareAlike 4.0 International License.

sbryan, 2015-07-15:
Awesome :+1: 

Marking as closed

jmkao, 2016-04-07:
Not sure if it warrants a re-open, but in setting up bintray releases, their automated system complains (although does not prevent) that CC licenses other than CC0 are not intended for software and sent me to read:

https://creativecommons.org/faq/#can-i-apply-a-creative-commons-license-to-software

The CC-BY-SA license is designed to be the the functional equivalent of GPLv3 and usable for documentation or artwork accompanying GPLv3 software, but lacks terms that would interoperate with other software licenses, like the BSD license that Microtome uses or the MIT license used by SLAcer.js, in addition to other drawbacks mentioned in that FAQ.

Practically speaking, this doesn't seem to be any impediment currently, but it may warrant discussion as we consider integrating with other projects.

WesGilster, 2016-04-08:
Yeah I agree. We may want to take a look at it sooner than later. I was thinking about integrating within the next couple of weeks.

jmkao, 2016-04-10:
I'll reopen this then since otherwise it doesn't show up on the default issue list.

There are some reasonable good and simple that break down the different licenses into a small number of families, basically permissive vs. copyleft:

http://choosealicense.com/
http://www.protecode.com/choosing-open-source-license-infogrpahic/

Looks like Microtome switched from the modified BSD to Apache 2.0 license last week, but functionally they're equivalent.

MIT, modified BSD, Apache 2.0, GPLv3 and LGPLv3 are all compatible with each other, so I think any of those would be fine and eliminate any issues with integration. The most similar to the CC-BY-SA would be GPLv3.

I personally like the simplicity and permissiveness of the MIT License, but really have no strong preference over any of the options so long as we get license compatibility to integrate.

jmkao, 2016-05-13:
@ergobot @WesGilster Any decision on this?

ergobot, 2016-05-13:
Users should always be able to view the source.  Contributor and community efforts to build this project should be open, transparent, and protected.  Those are my only requirements.  
From the links provided, I like the MIT and GPLv3.  I also like the simplicity and permissiveness of the MIT.  I like the protections of the GPLv3.
I'm for using the GPLv3 license.

jmkao, 2016-05-13:
GPLv3 sounds good to me; @WesGilster ?

kloknibor, 2016-05-14:
Both seem fine to me too ;)

WesGilster, 2016-05-14:
I'm good with GPLv3, but I'd like to make sure that we add author information on each of our source files. That gives the original author further privileges than is expressed in the GPL. **Don't worry, I don't have any plans of going on my own**, but I'm also not interested in losing rights due to a choice that I/we make.

jmkao, 2016-05-14:
I think that makes sense, it will also give contributors a sense of who to ask questions to regarding a given file. Do you think we should update our LICENSE file first and then incrementally add in author headers as we commit? Or do you want to hold back on the change to the LICENSE file until we have all the source code files squared away?

I personally lean towards the former, since git already has authorship information embedded in a file's history (well, so long as you `git log --follow` or go back to commits before the source tree change we discussed in #218) and for project integrations, we might want a software-oriented license sooner. (Also the pipe dream of attracting more contributors.) In theory it might even be possible to write a script to automatically create those.

WesGilster, 2016-05-14:
Sounds good, we don't have to wait for authors on source files. I don't think there is really that many source files that we can't "remember" where the authors came from.

jmkao, 2016-06-13:
I'm going to make the change now. Since we have additional committers coming on board, I want to make sure they are submitting to the right license.

commit 37771c03 (2016-06-13): Update LICENSE.md to GPLv3

Fixes #48
closed 2016-06-13 by jmkao