area515/Photonic3D #49: Error printer not started but can't stop any printer

kloknibor, 2015-07-08:
Hi,

When I start a printer and if I press a button before the printer is started completely, it gives an error that the printer isn't started. After this error the printer won't start at all.

Also when I try to restart the printer it gives the error that the display is already assigned to a printer and serial port. And I can't stop the named printer. The only solution is too reboot ;)!

Do you need some more information on this bug? (logs etc?)

WesGilster, 2015-07-08:
Correct me if I'm wrong, but it seems there are two separate bugs here.

To Reproduce Bug 1:
1. Start printer.
2. Immediately press "any button"
3. Receive "printer isn't started" message
4. Printer will never work from that point until CWH is restarted/OS 
rebooted.

To Reproduce Bug 2:
1. Restart printer. (Are you talking about a hardware procedure here. Or 
a CWH start and stop procedure?)
2. Receive "printer is already assigned to a printer and serial port".
3. Printer will never work from that point until CWH is restarted/OS 
rebooted.

Is this reproducible always? The logs would help solve the problem 
pretty quick. Also when you say "press a button" can you be more 
specific? Like which button, I understand it probably doesn't matter, 
but on the off chance I happen to pick a button that doesn't replicate 
this problem, it can get confusing.

Thanks,
Wes G.

kloknibor, 2015-07-08:
Hmm maybe I was to quick with reproducing this error... seems like if you wait long enough the printer would be "started" and work... It weren't 2 bugs, What I meant was indeed what you discribe but the bug 2 came out of bug 1. So when I tried to start the printer after bug 1 it gave me bug 2 ;) But I will test some more... sorry!

WesGilster, 2015-07-08:
No problem at all. I'm just going to wait for you on this. It's possible 
you are experiencing the following problem:
https://github.com/area515/Creation-Workshop-Host/issues/24

It's pretty tough to reproduce and I'm not sure I've ever got it to 
happen on the Raspberry Pi, but it could take 2 minutes for the printer 
to actually start due to the timeout. However the printer does actually 
start and operates normally after that time. If your interested in doing 
some testing, I created another Serial 
Implementation(org.area515.resinprinter.serial.JSSCCommPort) and I never 
really tried it out.

Obviously this is all speculation without your logs though.

Thanks,
Wes G.

kloknibor, 2015-07-09:
Well it was indeed just because I wasn't patient The errors only come up between these 2 minutes ;)! if I wait till the 2 minutes are over than everything is fine ;)! So yeah it has to do with issue #24 . I'm also using the new Draken printer and this would run on a modified version of sprinter firmware. Is it possible to speed up this process with this firmware?

Maybe make an popup to set different options ;)! like baudrate, firmware, machine and profile upload etc :)!

By the way I always used octoprint with my old FDM with marling firmware, didn't ever had problems ;) Maybe they found the solution!

WesGilster, 2015-07-09:
Cool, this is actually good news that you can apparently recreate this 
on demand.

Thanks,
Wes G.

WesGilster, 2015-07-09:
I'll reopen the #24 as a bug and start looking into why the 'ok' isn't 
sent back from the firmware.

Thanks,
Wes G.

WesGilster, 2015-07-14:
Closing since this has been deemed a duplicate of #24.

closed 2015-07-14 by WesGilster