area515/Photonic3D #50: Set default printer to start

sbryan, 2015-07-12:
I poked around the code and issues and didn't see anything that suggested this was even considered before, so treat this as an enhancement request...

I only have one printer, and I run CWH on an RPi2B that is "embedded" in it (mounted in the chassis, and powered by the same switch that powers the printer).  So the function is really pretty dedicated to this one device.

While I have multiple "printers" defined (for varying resolutions based on projector position), I would like to have one of them always "started" as part of the CWH initialization process.

I'd be happy with this just being a property in one of the `/root/3dPrinters/*.printer` files, like `<isDefault/>`, or, making it a cookie stored in the web interface local storage and a UI checkbox that offers something like "Automatically start the last started/selected printer every time?"

WesGilster, 2015-07-14:
Yeah cool idea. I put that in my next update.

WesGilster, 2015-07-19:
Add 'AutoStart' to your PrinterConfiguration in WesGilster repo cwh-0.113.zip.  (Untested)

sbryan, 2015-07-20:
@WesGilster, cool, thanks for working on this so quickly.  See my [comment](https://github.com/WesGilster/Creation-Workshop-Host/commit/9a06a2a1badb116782f82760d59c3ef841302267#commitcomment-12255438) in your repo/commit.  I'm not sure it actually works yet based on my quick code review, but I could be reading things wrong.  I've just returned from a weekend away and have not had the chance to test it yet, so please forgive my doubt w/out having actually tested...

WesGilster, 2015-08-10:
Fixed, Tested and Checked into CWH-0.114 (WesGilster repo)

WesGilster, 2015-09-22:
I know this feature is working.

closed 2015-09-22 by WesGilster