area515/Photonic3D #51: Update CWHost for 2015-05-05-raspbian-wheezy

SouthPaw-42, 2015-07-16:
Update CWHost for 2015-05-05-raspbian-wheezy and update instructions please

WesGilster, 2015-07-17:
In what way does CWH need to be updated for the latest version? Also, how do the instructions need to be changed?

SouthPaw-42, 2015-07-17:
When I tried to install CWHost I received errors. I will try install again and document the errors.
 I also can't find raspi-config-cwhost.sh anywhere after install.

WesGilster, 2015-07-19:
I don't believe we have ever installed raspi-config-cwhost.sh with the application. It's pretty experimental right now. Thanks for the help testing!

sbryan, 2015-07-20:
@SouthPaw-42, I've tested and debugged the raspi-config-cwhost.sh, and as @WesGilster indicates, it really is experimental.  I've written down a bunch of changes, and started making the changes necessary, but haven't had time yet to finish and submit a pull request.  I'll try to do that this week maybe...

For now though, I wouldn't even run it... it actually does more harm than good, and re-running it on a system it was already run on can make things worse.

WesGilster, 2015-08-10:
Marking this a duplicate of #47

closed 2015-08-10 by WesGilster