area515/Photonic3D #54: New GUI needs to be optimized for Mobile

WesGilster, 2015-08-23:
The new GUI works with mobile, but it really needs to have some optimizations to make it cool

WesGilster, 2015-09-07:
The specific bug I found was fixed in the latest version. I'm not sure this bug should be opened anymore. Need more testers

kloknibor, 2015-09-07:
Sorry my printer is still down... When I recieve finally the right parts I will dtart testing again and when all draken printers are fixed or sended out you will have some testers :)!

WesGilster, 2015-09-08:
No problem man I've still got to get these recent changes merged into 515.

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

Ergobot is working on a GUI that gracefully folds into mobile even better, but the current new gui works well enough.

closed 2015-12-27 by WesGilster