area515/Photonic3D #55: Make a version that pulls resources from CDN.

WesGilster, 2015-08-23:
At the moment the Raspberry Pi is hosting all resources and that puts quite a load on it. I could easily modify the build such that it you can run a version that downloads resources from a CDN. That way you could modify your config.properties like this:
hostGUI=resourcesCDN

...and it would take quite a bit of load off the Raspberry Pi. Keep in mind, this would force all of your CWH clients to have Internet access... It's a tradeoff

DanielJoyce, 2016-11-28:
Eh, if the proper cache headers are being sent this won't be an issue after the first browser load.

WesGilster, 2016-11-28:
Yeah, I've had second thoughts about this as well.

kloknibor, 2017-07-11:
Shall we close this then? I agree that with cache this should be sufficient  + it allows the user to only have an internal local network connection without acces to he internet

WesGilster, 2017-07-12:
Yeah it's awfully pointless...

closed 2017-07-12 by WesGilster