area515/Photonic3D #58: Projector Model Detection Framework

WesGilster, 2015-09-21:
Detects both the presence and model of the projector.

Implemented and untested in version 0.121.

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

Working very well. I'm pretty excited to start adding more printer models for auto detection...

nerginer, 2016-02-14:
Hi,
I have vivitek D535 in my hand and trying to use it with CWH. I can on and off my projector from terminal but I can not set the config.properties

I try the below sentence but CWH can not detect it. Please find the attached projector communication doc.

[RS232_codes.pdf](https://github.com/area515/Creation-Workshop-Host/files/129785/RS232_codes.pdf)
 

hexCodeBasedProjectors=[{"name":"vivitekD535","onHex":"7Eh50h4Eh0Dh", "offHex":"7Eh50h46h0Dh","detectionHex":"7Eh71h56h0Dh", "detectionResponseRegex":"4450333430362d585558582d4450503031"}

WesGilster, 2016-02-14:
I think there are a few things going on here:
1. You are missing a ']' at the end of your string
2. Your hex strings have unnecessary 'h's inserted into the strings. These will fail the parse since an h is not hex grammer(0-9a-fA-F).
3. While I can't say that your detection response string is incorrect, I am pretty skeptical. Regular expressions are not hex based, they are ASCII. So if you are interested in using hex strings, you need to do what some of the other reg ex strings are doing by escaping them like \\\\x0B (or whatever the hex code is).

However, if you really want to use ascii(which is what it seems you are doing) then you are doing it perfectly. Here are the reasons I was a bit skeptical:
1. Outside of the 'd's in your regex, it seems as though you are attempting to specify decimal notation which isn't going to work.
2. I'm curious about the 'd's I see embedded in your response string. Are those attempting to be carriage returns? If so, that won't work.
3. Be careful that many projectors like to return the state/mode/extra info about the projector when you provide the power on code. When this happens, the detection string can vary greatly. That is precisely why I designed this to be a regular expression and not a very precise value like you've provided.

Much like most serial port projector documentation, the responses from your projector are not well documented. I'm assuming that you got this information directly through some tool. You'll need to know what format it gave you this information in. If you are unsure, I've built a tool with the hardware test kit that helps you do this if you are interested in it. It will provide you hex strings that you will need to escape as I described above. It's also handy because you can copy and paste hex strings(without quoting) directly from your documentation and get responses. Let me know if you are interested.

nerginer, 2016-02-14:
Thanks for the answer. I have to tell you what I did to build this sentence first. It can be silly.
I copy and paste the on and off commands from the document. They are like

**Operation**
Power On
**ASCII**
~ PNCR
**HEX**
7Eh 50h 4Eh 0Dh

**Operation**
Power Off
**ASCII**
~ PFCR
**HEX** 
7Eh 50h 46h 0Dh 

I am sending ASCII from the terminal and can on and off the projector.
my first question is can I also write ASCII inside the quotes?  

For the auto detection I can not find any query inside the projector document so I choose to query the software version. I got the response from the terminal and convert it to hex. 
I also read that
※ Power On command will only word at system standby. 
so how auto detection framework can detect an off projector. Do I have to make it on before starting the CWH?

Last question is Can I disable the auto detection mode and just send on and off commands? I want to add the command at the end of the g-codes so that the projector will turn off after the completion of the print job. It can be hard to handle all  projector models so it can be useful to disable the auto detection.

Thanks for the help and this great software. I will be more than happy to test it and give you feedback

Regards
Nuri

WesGilster, 2016-02-14:
> can I also write ASCII inside the quotes?
regex responses=yes if you are sure you know the response in ASCII.(hex must be escaped)
HEX requests=no. (Only hex is available for these)

> Do I have to make it on before starting the CWH?
No, sending the "hex on" in the detection regex will work nicely. My printer is the same way. Be aware that you may get different responses depending on the state of the projector(stand by/off/on). It's up to you to determine what those are, but it seems you've already started that.

> It can be hard to handle all projector models so it can be useful to disable the auto detection
This is partially true, but I'm not yet convinced that it's significantly more than it would take to support a new projector anyway. I agree that there will be an upfront cost to perform autodetection, and you and I will end up being the one that bears that cost for this projector. On the other hand, the benefit to autodetection is huge. Things just simply work. In the world of 3d printing(especially resin printing), that offset's this small upfront cost...

> Last question is Can I disable the auto detection mode and just send on and off commands?
This isn't supported since it already knows when to perform these things at the appropriate times. Think about this from an end user standpoint. They really shouldn't be thinking about when to perform these things. Your/our work is deciding on how to detect the projector.

Most importantly, thank you for testing! We'll be interested in your printer profiles and projector codes when you get things up and running.

WesGilster, 2016-02-14:
Sorry didn't even get to your codes. This should work for on/off codes:

hexCodeBasedProjectors=[{"name":"vivitekD535","onHex":"7E504E0D", "offHex":"7E50460D","detectionHex":"7E71560D", "detectionResponseRegex":"?"}]

However, I still haven't determined how you are coming up with this string:
4450333430362d585558582d4450503031

nerginer, 2016-02-14:
Ok here where I am
When I send on command the projector power on and response nothing.
below I write power on code as detectionHex and this powered on the projector but CWH can not determine the projector. 

We are very near the solution 

#hexCodeBasedProjectors=[{"name":"vivitekD535","onHex":"7E504E0D", "offHex":"7E50460D","detectionHex":"7E504E0D", "detectionResponseRegex":""}]

nerginer, 2016-02-14:
I want to query the software version as auto detection. It is 7E71560D I am getting DP3406-XUXX-DPP01 when I send this from the terminal so I build up the hexCodeBasedProjectors as below. It did not work. I try it while the projector is on but no way. The projector is only responding to power on command.

I am not sure if your auto detection code can work with vivtek projectors.

hexCodeBasedProjectors=[{"name":"vivitekD535","onHex":"7E504E0D", "offHex":"7E50460D","detectionHex":"7E71560D", "detectionResponseRegex":"DP3406-XUXX-DPP01"}]

WesGilster, 2016-02-14:
vivtek is just a brand name. What matters is just serial input/output and a little bit of work.

Could you try something like this?
hexCodeBasedProjectors=[{"name":"vivitekD535","onHex":"7E504E0D", "offHex":"7E50460D","detectionHex":"7E71560D", "detectionResponseRegex":"(?s:.*DP3406.*\\\\x0D?)"}]

nerginer, 2016-02-14:
Did not worked. I try it both with  on and off state 
Are you sending onHex  before detectionHex programatically?

WesGilster, 2016-02-14:
Are you running on Linux?

WesGilster, 2016-02-14:
I've noticed that I didn't properly put the above string where you can copy it from github. I've fixed it in the above comments and in these below. The asterisk wasn't properly shown.

`hexCodeBasedProjectors=[{"name":"vivitekD535","onHex":"7E504E0D", "offHex":"7E50460D","detectionHex":"7E71560D", "detectionResponseRegex":"(?s:.*DP3406.*\\x0D?)"}]`

WesGilster, 2016-02-14:
Crud, the backslashes are still not showing up properly.
The best option is my projector tool. 
If you are running on Linux, you can run it like this:
```
sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/testKitDev.sh
sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
sudo chmod 777 *.sh
sudo ./testKitDev.sh
```

Afterwards, you can run this:
`cd /opt/cwhTestKit`
`sudo java -Djava.library.path=/usr/lib/jni:os/Linux/${cpu} -cp lib/*:. org.area515.resinprinter.projector.ProjectorOutput`

WesGilster, 2016-02-14:
You will be taken through a series of screens that look like this:
0. /dev/ttyACM0
1. /dev/ttyUSB0
Type the number of the Serial Port that you would like to test (then press enter):
1
0. 9600
1. 115200
Type the number of the speed that you would like to test (then press enter):
1
0. Acer
1. ViewSonic (pjd7820hd)
Type the number of the projector that you would like to test (then press enter):
1
0. On Hex
1. Off Hex
2. Detection Hex
3. Custom (0x00 0x00...)
Type the number of the hexcode that you would like to test (then press enter):
3
Enter custom hex string in the form "0x00 0x00 0x00" (then press enter):

WesGilster, 2016-02-14:
The printer you choose doesn't matter since you will be entering your own custom (0x00 0x00) codes.

If this is still too painful. I can write the program that gathers all of this information for you and you can just run it. It would probably be good to have this program for on-boarding new projectors anyway...

nerginer, 2016-02-14:
Windows at the moment.

nerginer, 2016-02-14:
I can test it tomorrow with Rasppi and let you know.

Thanks alot for your support.

WesGilster, 2016-02-14:
Did you try it with the four backslashes instead of two? That should work.

nerginer, 2016-02-14:
WOW four backslashes worked when the printer is on

nerginer, 2016-02-14:
is it possible to send on command at the beginning. 
And what is the code to enter at the end of the gcode to make it off

WesGilster, 2016-02-14:
> WOW four backslashes worked when the printer is on

:) That's good news. I assume you mean when the projector is on, not the printer.

> is it possible to send on command at the beginning. 

Yes, in fact you can already do that by the following detection hex:
7E504E0D7E71560D
You can try this out, however I'm awfully concerned it isn't going to work. I'm going to need to put a delay between those two hex codes. Since your documentation mentioned this exact phrase:
> Minimum delay for next command: 1ms 

Java is going to be pushing data much faster than 1ms, and your projector probably won't catch the second command. That means I'll need to change our hex strings to be based upon a freemarker template instead of simple hex. Then what you will do is write something like the following:
7E504E0D{delay(1)}7E71560D
I'll try to have something out tomorrow, but I'm still trying to get the live streaming working, so now promises...

> And what is the code to enter at the end of the gcode to make it off
CWH is smart enough to know when your print is starting and when it is complete. It is designed to know the appropriate times to turn on your projector and when to shut it off when the print is complete. If you find it's not performing these functions appropriately, that is a fairly simple bug that we need to fix. 

I'm also playing around with the idea of shutting off the projector when the print is paused, I'd be interested in getting people's opinions.

jmkao, 2016-02-15:
I think shutting off the projecting during a pause would only work for printers that have shutters. Most projectors have unstoppable splash screens that would have dire unintended consequences on the paused print.

WesGilster, 2016-02-15:
@jmkao
That's a really good point. I don't think I'll fiddle with that...

@nerginer 
I fixed the live streaming issue I was having, so I should be able to fix this tonight. Although I'm wondering if it's necessary now. Did you check to see if you could query the software version while in standby mode? You can use hyperterminal or whatever program you were testing this with before, you do don't need to use CWH to determine this.

Both of us have read the following in the projector model:
> Power On command will only word at system standby. 

That means 1 of two things:
1. 'Power On' hex will only execute while the projector is in standby mode.
2. 'Power On' hex is the only command that will execute while the projector is in standby mode.

I took it to mean 1...

nerginer, 2016-02-15:
7E504E0D{delay(1)}7E71560D is what I need. If we will communicate with different brand of projectors we will need that delay function.

WesGilster, 2016-02-15:
Alright, I should be able to put something together tonight. I won't be able to use Freemarker for this since the parsing phase with absorb the delay. I think I'll go with an embedded javascript solution instead.

nerginer, 2016-02-15:
great to hear that
thanks

WesGilster, 2016-02-16:
This is implemented in cwh0.228 in my development branch. I went with the embedded javascript solution since Freemarker wouldn't work. The parsing isn't very sophisticated, but it will work for simple javascript function declaration and execution.

7E504E0D(function (){java.lang.Thread.sleep(1);})()7E71560D

After you get this working, I'd also like to ask you to give me the output from your bulb hours hex:
7E714C0D

and make sure that the return value is the same as what your projector's bulb hours menu option says.

nerginer, 2016-02-16:
Thanks Wes,

I want to be the part of the development team. I have some ideas about developing some electronics for CWH. I can also run some tests and help for the wikis. Can you please add me as a contributor? 
Does your development branch github page is https://github.com/WesGilster/Creation-Workshop-Host 
I am building DLP hardware that I have my own deigns. Being the part of the team will be great.
Hope to test and give you feedback about Vivitek (D535) code you added in a day. You are great 

Regards,
Nuri

WesGilster, 2016-02-16:
That sounds great. We'll be sure to get you added, but that's mainly to modify the wiki pages and stuff. The best way to get code added is to fork my repo and do pull requests to me. The reason we do it that way is because everyone's installation of CWH is automatically pulling from the prod repo(area515) and we are trying to keep that branch stable. If you are interested in installing from anyone's repo you can do it with the start script that comes with CHW.

nerginer, 2016-02-16:
Of course I will get the dev branch from your repo and make pull requests to you. 
In the other hand; I want to use all the hardware possibilities that CWH supports. So that I can design a board with all supported hardware. Can you please tell me more about the supported hardware list and the near future implementations? I mean the hardware like the servo or resin level indicator. 
I have an idea to develop a uv sensor probe hardware. We can work together and the system can generate the projector mask automatically by sampling the uv light radiation on the tray. Is this the suitable place to talk about these subjects? I do not want to mess the issues on github. 
Thanks alot for your cooperative approach.

kloknibor, 2016-02-16:
I'm also currently developing an UV sensor probe... We have some iterations but we are trying to reduce costs currently. Electronics are already ordered, however these are all standard components. I did have 2 electronic engineering students in my team but they are unavailable for an while... (because of school). And I'm planning to make an windows app at first (already developing in C#) and I also talked with Wes about integration with CWH. To be honest I wanted to keep it quiet that I was developing this and than bring it to the market. Also I'm not planning to make this device opensource. But maybe we can help each other out... My team is namely best at mechanical engineering and out of the box thinking, but I don't have the programmer and electronics capability that I would like to have... 

Drop me an e-mail at info@drakenprint.nl if intrested ;)! 

Ps : Our plans stretch beyond only an UV probe ;)! But we are a startup...

nerginer, 2016-02-16:
Wes,
I get the dev branch and test it with my vivitek D535. It is working great. Thanks for implementing the delay function. For the lamp hour I query the projector and I got 26. I can not confirm with the menu because my vivitek is optically modified and it is nearly impossible to use the menu. But I assume it is correct. You can push this part of the code to the production. Great job.

Regards,
Nuri

WesGilster, 2016-02-17:
Wow that was quick, usually I have to wait a while for confirmation of the fix... 

To your question about projector functionality:
The projector hours I requested are designed for adjusting exposure times. While you could use them to generate a rough projector mask in the projector mask calculator, the projector hours weren't designed for this purpose. For example, I used the default settings for the mUVe1 for some of my first prints and melted my model directly to the build vat silicone during the widest part of the print. JMKao has also mentioned 30% longer exposure times as he reaches the 250 hour mark of his projector.

You already heard from Kloknibor about the UV sensor, I plan on integrating with his hardware for automatic generation of projector masks. 

I'm not sure I can name all of the hardware that we support right now, we are starting to get a little more organized in that area. To name a few WIFI, raspicam, several firmware, several projectors and several printers. If there is a piece of hardware that you would like to integrate with, send me the specs for the hardware. If I think I can tackle it, I'll ask you to send me the hardware. 

We do have some limits when dealing with Raspberry Pi GPIO and control of realtime hardware with precise timing, but otherwise the sky is the limit...

If you are looking at integrating printer hardware directly, it's pretty easy. Just extend our improperly named: org.area515.resinprinter.gcode.GCodeControl class and then set that classname in your printer profile. Right now we have 2 implementations:
SedgwickGCodeControl: (which doesn't use gcode)
eGENERICGCodeControl: (which does use gcode)

closed 2016-02-17 by WesGilster