area515/Photonic3D #61: Crashing upon making a printer

kloknibor, 2015-09-28:
Hi Wes!,

I'm finally able to test properly ;)! but cwh already crashes upan adding a printer... I get a 404 message... logs can be found here : https://www.dropbox.com/sh/cl51eienirb6dh6/AADP1Aa442Xqh_3GTgJnx1ema?dl=0 (yeah yeah I know how to get logs now :P) hope that helpt ;) if you need anything more please ask :)

thanks in advance!

Robin

Ps it was with the new GUI enabled and my mail settings also filled in, fresh install other than that newest raspbian updated upgraded and configured ;)!

WesGilster, 2015-09-28:
Your using the area515 repo of CWH. That's much too old to work under the new gui. Run this:
cd /opt/cwh
sudo ./start.sh WesGilster

WesGilster, 2015-09-29:
I reproduced this bug and fixed in CWH-0.134.

kloknibor, 2015-09-29:
I will have to wait till tomorrow to test... But thanks for the quick response! And one question left, if I run the WesGilster beta version once, Does this start by default with the raspberry pi?

WesGilster, 2015-09-29:
No, it will attempt to upgrade against area515, and will decide a version isn't available since my version will always be higher.

CWH will always attempt to use and upgrade to the most recent version and it will never go backwards unless you force the downgrade with:
cd /opt/cwh
sudo ./downgrade.sh
    which is nothing more than this: ./start.sh area515 force

So, for example, if you are on a supported 3d printer platform, you would do this to go back to your printer manufacturers stable version:
   sudo ./start.sh [manufacturersRepo] force

If you are interested in seeing what is happening behind the scenes, you can determine if an upgrade will happen by checking your current version against the version number corresponding to the Repo you specified on the command line:
https://github.com/[SomeRepo]/Creation-Workshop-Host/blob/master/host/cwh-0.XXX.zip

If you didn't specify a Repo on the command line, it will use area515 by default.

kloknibor, 2015-09-29:
Thanks for the explanation! I get it now ;)!

WesGilster, 2015-10-03:
Since ergobot merged the last build this is fixed in area515 as well.

WesGilster, 2015-11-06:
Fixed a while back...

closed 2015-11-06 by WesGilster