area515/Photonic3D #62: Send Gcode at startup printer

kloknibor, 2015-09-29:
Hi,

My setup is as following. I've got a DLP printer and my raspberry pi gets his power from the printer itself. So my beamer is just off/in standby. I can send a Gcode command (M605 to turn the projector on and M606 to turn the projector off). But to achieve this I currently should start a printer with a simulated display send the Gcode to turn on the projector so this way there is a display available for the real printer, turn off this printer and start the real printer ;) 

Would it be possible to add custom Gcode at startup of a printer and stopping a printer or would there be another way to just turn on the projector so I can start the printer with a lot less hassle?

thanks in advance

Robin

WesGilster, 2015-09-29:
Yeah that would be pretty easy to do, I'll have to think about whether 
that should be done via gcode though. I'm trying to build an abstraction 
away from gcode, so that printer manufacturers don't "Send gcode" 
instead they will perform a series of startup functions that may include 
projector power and other tasks, this will allow manufactures to define 
gcode for an action one time instead of adding it to several different 
places like "Printer startup gcode" "Printer standby gcode" etc...

Thanks,
Wes G.

WesGilster, 2016-01-01:
I was just reading this again. I can add this functionality pretty quickly, but I'm wondering if it's necessary. If you want to send gcode, you can just send them in your gcode header/footer print sections. On the other hand you should probably use the direct projector communication which is now setup in the config.properties. This will allow you to automatically find your projector and CWH will handle powering it off/on at the appropriate times.

WesGilster, 2016-01-17:
Projector autodetection was pretty sloppy until cwh-0.200 or so. There should be no manual projector interaction now.

closed 2016-01-17 by WesGilster