area515/Photonic3D #63: No facility to set base layer exposure time

ps-bond, 2015-09-29:
Per title - there appears to be no way to set a base layer exposure.

WesGilster, 2015-09-29:
There are a ton of ways to set layer exposure time, but not through the UI. Yet. You can use the settings that you migrated from CW, or my favorite you can use a Javascript calculator to base it off of the currently exposed surface area like this:
    <ZLiftDistanceCalculator>var value = 9.0;
if ($CURSLICE &gt; $NumFirstLayers) {
 value = 3.5555555555555420e+000 * Math.pow($buildAreaMM,0) + 4.3333333333334060e-003 * Math.pow($buildAreaMM,1) + 1.1111111111110492e-006 * Math.pow($buildAreaMM,2);
}
value</ZLiftDistanceCalculator>
    <ZLiftSpeedCalculator>var value = 0.25;
if ($CURSLICE &gt; $NumFirstLayers) {
 value = 4.6666666666666705e+000 * Math.pow($buildAreaMM,0) + -7.0000000000000184e-003 * Math.pow($buildAreaMM,1) + 3.3333333333333490e-006 * Math.pow($buildAreaMM,2);
}
value</ZLiftSpeedCalculator>

WesGilster, 2016-01-03:
Fixed in cwh-0.183

WesGilster, 2016-01-17:
This was fixed long ago, and the writeup was pretty impressive:
https://github.com/area515/Creation-Workshop-Host/wiki/Slicing-With-Creation-Workshop-Host

closed 2016-01-17 by WesGilster