area515/Photonic3D #64: User documentation lacking

ps-bond, 2015-09-29:
Unclear what format is expected by the code, no usage instructions.

WesGilster, 2015-09-29:
Yep your right. Not sure this is going too happen soon. CWH is probably targeted closer towards 3d Print Manufacturers than end users. By the time the public gui is complete, it's going to be pretty vast and won't keep up with the functionality. For me the functionality is more important than the UI. If a printer manufacturer is interested in a custom user interface, we don't mind building that for a cost of course.

File types currently supported:
1. .stl
2. .mazecube
3. .cws/.zip

File types supported in near future:
1. .jpg/.gif
2. .txt

ps-bond, 2015-09-30:
I know documentation is the least glamorous section of dev, but if it's not there the code can't really be used.  I'd happily help write some of it, but since ATM I don't know how it's supposed to be used it's a tad tricky to tell others.

WesGilster, 2015-09-30:
Let me know what documentation you are interested in seeing and I'll see if I can help you out.

I'm not really sure the Raspberry Pi is the best hardware for the job either, which is why I haven't written any code or script to support anything that has to do with it. Even the start script doesn't have anything to do with the Raspberry Pi; in fact, the scripts don't even depend on the Raspbian OS. I use CWH on a laptop which is Ubuntu based.

That being said, that hasn't stopped print manufacturers from working with us to to ensure it works on the Raspberry Pi. Keep in mind, the same is often said of the Arduino board (we all have heard the Arduino called a prototyping board), but that isn't stopping print manufacturers from using it in production environments. I suppose I could say that quadcore CPU, 4 USB ports, increased power stability and hdmi are more than sufficient than a questionable Windows tablet market, but CWH runs on both of these(and more) so the conversation really doesn't mean much.

I guess I should be fully accurate. I did support the Raspberry Pi camera because it was donated and it was easy to add the following to a config.properties file:
streamingCommand=raspivid -ih -w {0} -h {1} -n -t 0 -o -

Is there some production ready hardware you were interested in targeting?

ps-bond, 2015-10-01:
An idiot-level guide to initial setup would be a good start - per the first line it's not even clear what filetypes are expected.  Something like a 10-step setup guide to go with the install guide, perhaps?

Interesting - all I've seen of CWH has portrayed it as a Pi solution (or at least a small embedded device solution avoiding the need for a tethered laptop).  With Arduino, PICs et al in production environments (IME) the board is usually respun so things like shields are incorporated into the design so there's no separate plug-in modules.  20 years as an embedded sw engineer until I decided to switch to making jewellery full-time last year (hence 3d printers); C, C++ & Python were/are my primary languages, never had to sit down & use Java in anger but if it helps I'll learn.

So no poduction hw, sorry :)

WesGilster, 2015-10-01:
Cool thanks for you willingness to dig in. You already know that syntactically moving from C++ to Java isn't really the hard part. The only real learning curve is the vast API's available and how to use them rather than build your own.

How about a compromise. We work diligently to get your system up and working with all the features, and we quickly release fixes for each of the bugs that you run into. Could you then be in a position to write this guide? That way, we have this documented from the perspective of of an outsider. I'm guessing you are already pretty close to an operational system, so it might not be much of an effort.

WesGilster, 2015-10-01:
It seems you had another question as well about file type formats:
1. .stl
This should be self explanatory(they can be ASCII or binary) just download it or give the link to CWH and have it download it for you.
2. .mazecube
Documented here: #52 
3. .cws/.zip
Video here: https://www.youtube.com/watch?v=J3HTCkxlKcw

kloknibor, 2015-10-21:
I wrote a manual by the installaton video I made, maybe you find it usefull enough ;)! here it is : 

Step 1 :

Download SD card formatter and Win32diskimager from :

SD card formatter : https://www.sdcard.org/downloads/formatter_4/

Win32Diskimager : http://sourceforge.net/projects/win32diskimager/files/Archive/

Step 2: 

Install SD card formatter and Win32diskimager

Step 3:

Insert the (micro)SD card you want to use with your pi and start SD card formatter. Make sure the right device is selected.

Also click option and set "format size adjustment" to ON.

Click format and OK.

Step 4:

Download raspbian "Wheezy" from here : https://www.raspberrypi.org/downloads/raspbian/

Step 5: 

Start Win32diskimager and select the .img file you downloaded in step 4. Also make sure the correct device is selected and click write > Yes. This can take a while ;)!

Step 6:

After Win32diskimager says it's finished disconnect your sd card safely and put it in your raspberry pi. Connect the pi to your network and power up your pi ;)! (In this tutorial we use an ethernet cable but this could als be by a WiFi network But you first need to setup your wifi, here you can find how to do that (if drivers work out of the box) : https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md)

Step 7:

Download advanced ip scanner from here : http://www.advanced-ip-scanner.com/nl/

Step 8:

Start advanced ip scanner and scan your network. Find the ip your raspberry pi is using and note it somewhere. 

Step 9:

Download putty from here : http://the.earth.li/~sgtatham/putty/latest/x86/putty.exe

Step 10:

Start putty and enter the ip you found in step 7 and connect. You get an login screen enter "pi" as username and "raspberry" as password.

Step 11:

After login enter sudo raspi-config and expand file system and enable camera. After this click finish and choose reboot.

Step 12:

reconnect putty as in step 10

Step 13:

run/paste the following code in putty :

sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
sudo chmod 777 start.sh
sudo ./start.sh

Yayy you installed Creation workshop host on your raspberry pi ;)! You can access it from theipofyourpi:9091 or see optional:

Optional :

Download cwhClient-x.xxx.zip (the x are numbers) from https://github.com/area515/Creation-Workshop-Host/tree/master/host and unpack it. 

after unpackin run browseprinter.bat and it will open up your printer ;)

ps-bond, 2016-01-21:
Right, apologies for the radio silence - Christmas is entertaining in the jewellery world.  

Back and started looking at the issue again; current use case is:

CWH v.202

RPi attached to LittleRP; speed has been altered to 9600 in /root/Machines/<whatever I called the thing>
Projector is an Acer H6510BD; it has a 9 pin serial port (as well as a mini USB connector, but that clearly does not have a serial impl behind it).  Not connected to the Pi ATM; I have a suitable USB/RS232 adapter but can't currently locate my gender changers - haven't used them in about a decade.  So, no serial connection at the moment.

Printer serial port set to /dev/ttyUSB1, display to 0:0.  Printer starts - the display goes black - then under controls I've failed to succeed in displaying a calibration screen or moving the Z axis.  

Feels like it's close but I'm missing something.

WesGilster, 2016-01-21:
Many firmware must be z homed before they are able to use the z axis. Without performing that event, you won't be able to use the z axis controls. As to the display, that's strange, because if it turns black that's usually a sign that I've taken it under control, and if it's under control, I've never seen it not work.

What we need is a log bundle. You can set that up in the email section of the the Settings tab. You can use my email address or your own directly in the 'serviceEmailAddresses' section. If you use your own, you can just drag the bundle directly into this comment section.

WesGilster, 2016-01-21:
Oh, and you can send a log bundle by clicking the "Send support info" link that is on every page of CWH.

ergobot, 2016-01-21:
I'll try on my littlerp this weekend to replicate the problem.  I'm using a
pi 2, wheezy.  What type of pi are you using?

ps-bond, 2016-01-21:
Pi2 as well.

Homing the LittleRP in sw isn't something I've had to do (explicitly) using CW; as there's no position feedback it has seemingly assumed wherever it starts is home.

Tried using the send support info fn (via gmail smtp) and Gmail has rejected it and flagged me with "Someone just tried to sign in to your Google Account ...@gmail.com from an app that doesn't meet modern security standards.".  How helpful of them.

Which are the relevant files for the log bundle?  log.err & log.out are the obvious ones.

kloknibor, 2016-01-21:
go here : https://www.google.com/settings/security/lesssecureapps and click enable ;)! Now mail by gmail works :)

ps-bond, 2016-01-22:
Somewhat bizarre - the lack of cal screen has now resolved.  Haven't rebooted the Pi, but I did power down the projector overnight.  Other than that...  No changes.  Might try a reboot & see if it reproduces, but other than that it may need to be filed as PEBKAC.

Still no life out of the motors; the device is definitely there under lsusb at least.

ergobot, 2016-01-22:
[image]
[image]
[image]
[logout.txt](https://github.com/area515/Creation-Workshop-Host/files/100890/logout.txt)

Using littlerp, grbl, pi2.  No projector connected to serial.  

Attached are log.out (logout.txt) and three gui screenshots.

ps-bond, 2016-01-22:
Progress.

1)  I've reproduced the issue with not being able to display the cal screen.  From the cal screen displaying/blanking ok, stopped the printer, rebooted CWH by running start.sh then starting the printer.  Display went black, but did not then display cal when instructed.  Rebooted pi.  Less inclined to suggest PEBKAC on this one now!
2)  Noticed that the /root/Machines/LittleRP.machine serial comms section was out-of-step with what was being set in the GUI; removed the printer via GUI (NB - removes LittleRP.slicing but not LittleRP.machine or LittleRP.profile; assumed these were throwbacks to an earlier version so deleted them).
3)  Recreated LittleRP via GUI; finally noticed the checkbox beside the serial port selection - which of course offers 9600/8/n/1...  
4)  Started with console output set for the projector, exception thrown & error displayed as above.
5)  Started with projector set to "select serial port", printer starts, log file shows comms with printer (incl.   printer open) and - once enabled - z axis moves.  

Next step is to config lift & exposure vals & give it a suitable file to work with.

I am writing notes as I go as well...  I'll do a full clean install (OS & CWH) when I run through the sequence prior to actually writing it up in the wiki.

WesGilster, 2016-01-22:
There is so much going on with this thread, that we should probably start separating these issues.

To Ergobot and ps-bond:
"Console Testing" should have been called "GCode Response Simulation" and is only appropriate for substituting out 3d printing firmware. The way that it works is that it completely circumvents talking to your 3d firmware and simply prints to the console. It's only a good option if you don't have any firmware connected to CWH and you still want to test the GUI. Using it for your projector is all sorts of wrong. When the autodetection sequence is executed against the simulation the "Console Testing" will return "OK" and won't make any sense. I'll rename this option to be more descriptive of what is actually being done, and place a sanity validation to keep people from doing this.

To ps-bond:
Awesome that you've recreated the issue. Is it possible to recreate on demand? If so, could you describe the sequence further. I'm a little slow...

To All:
Wiping out CWH isn't going touch your printer profiles and it also won't affect your overridden config.properties from the GUI either. This allows us to be very resilient against upgrades, but if you have some old profiles that aren't compatible, you could have some trouble as ps-bond just did. 

To ps-bond:
You asked previously which files are important in the log bundle, but they all are. For example, I could have seen that you were using console testing inside of your active printer profile, because it zips your config files as well.

WesGilster, 2016-06-23:
I renamed the GCode Response Simulation long ago and is in production. Autodetection is also been fixed months ago and documentation for how this process works should probably be tracked under bug #208. Reopen this bug if you want it tracked here.

closed 2016-06-23 by WesGilster