area515/Photonic3D #65: No debug output

ps-bond, 2015-09-29:
Running the code in debug mode - or even being more verbose in /var/log/messages would make it somewhat easier to understand when things go wrong.

WesGilster, 2015-09-29:
It seems this will probably happen a little earlier than what I had hoped.  Essentially you'll be able to change the loging output with log4j soon enough.

WesGilster, 2016-01-24:
log4j has been implemented in the dev version cwh-0.209. This was a huge upgrade, that most people probably won't even be able to respect.

You can see the configurations available, without having to write single line of code:
http://logging.apache.org/log4j/2.x/manual/

@ps-bond You wanted to be able to log to /var/log/messages. The following configuration should allow you to do that: http://logging.apache.org/log4j/2.x/manual/appenders.html#SyslogAppender

There are a few features left that we may want to look into for the future:
1. Enable "on the fly" logger changes through the web.
2. Persistent log changes across upgrades.
3. Is the deletion policy configuration working on the old archived files? I don't think so and I don't want to fill up the Raspberry Pi. 
4. Someone might want to take a look at the policies I made in our log configuration. I used the properties style configuration, but you can use any format(properties, json, xml) to configure. The current file is called:
log4j2.properties

If you decide to change the name of our log configuration file, there are a few places you'll need to change:
1. Log file bundling in MachineService
2. start.sh and start.bat for startup functions
3. launcher configuration for Eclipse
4. If you change the format/file type you'll need to add them to our build configuration file.

The other good news is that on top of the performance gains that are inherent in log4j2, there is less object creation, less String concatenation and less unnecessary code execution(closures) than what regular sysout can provide.

closed 2016-03-11 by WesGilster