area515/Photonic3D #66: On-screen slice render intermittently stops working

ps-bond, 2015-09-29:
Slicing a test STL, the on-screen display (is this the console output referenced?) stops displaying any slices.

WesGilster, 2015-09-30:
The term "Console Output" means that it won't actually send the gcode to the printer, instead it sends the gcode directly to sysout which is currently redirected to your logs: /opt/cwh/log.out. 

I'll need your logs to determine why it stopped working. Keep in mind, if you've shutoff CWH since you experienced the problem, we've lost the ability to troubleshoot this. Is this easily repeatable? Someday I'll build a log bundling facility or something else to make this easier. Maybe just use an email appender for log4j or something like that.

WesGilster, 2015-09-30:
Keep in mind, you can drag and drop files directly into this comment section so we can see them right here in Github.

ps-bond, 2015-09-30:
So there *is* debug output?  OK, that's useful to know.

I'll have a mooch around the FS, probably later today, and see if the log files still show anything; neither the Pi nor CWH have been shutdown.  The test print completed but never showed slices (or updated the slice count unless the print was paused).

WesGilster, 2015-09-30:
Ah, now that you explain further I'll bet I know what happened. Is it possible that you closed down, or left the browser? Our notification framework uses web sockets so that all clients watching the printer are notified when an event happens. However, if you closed down or left the browser, the web socket won't attempt to reattach since it doesn't know which printer to "attach" to. I probably won't fix that in the Old gui. The new gui will automatically "attach" to a printer when you browse to it.

ps-bond, 2015-10-01:
Left the browser?  I switched between tabs, but not shut it down.
I'll run it again and see what happens from boot.

WesGilster, 2015-10-20:
Sorry I messed this up pretty bad in the "stable" area515 build. Here's what I've done to fix it, and hopefully the tests will prevent this from happening again in the future. I'll post this in a couple of other places...

Use:
sudo ./startdev.sh 
to get the latest version of CWH from development until we can get the area515 build up-to-date.

Fixed in cwh-0.154
Rewrote EOL parsing and welcome chitchat parsing
Fixed NPE on close issue
Fixed automatic firmware detection on printer start
Fixed tests to perform firmware detection and 3dfirmware serial startup 10 times in a row
Fixed "can't start printer after it started once"
Optimized performance for JSSCCommPort thereby breaking RXTXCommPort
Changed default SerialPort implementation to JSSCCommPort
I'm starting to really dislike the RXTX implementation

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

Problem was with unstable websockets. I built a recovery mechanism so that the web sockets will reconnect after no activity or other problems.

closed 2015-12-27 by WesGilster