area515/Photonic3D #67: No facility to set layer height when slicing STL

ps-bond, 2015-09-29:
Per title.  Not clear what layer height is being used.

WesGilster, 2015-09-29:
The new gui will eventually allow you to set all of the STL slicing options, but until then you'll need to settle with the options you've migrated from your .slicing file from CW.

    <InkConfig>
     <PercentageOfPrintMaterialConsideredEmpty>0.0</PercentageOfPrintMaterialConsideredEmpty>
        <Name>Default</Name>
        <SliceHeight>0.1</SliceHeight>
        <LayerTime>8000</LayerTime>
        <FirstLayerTime>20000</FirstLayerTime>
        <NumberofBottomLayers>10</NumberofBottomLayers>
        <ResinPriceL>65.0</ResinPriceL>
    </InkConfig>

WesGilster, 2015-09-29:
Sorry reopening.

ps-bond, 2015-09-30:
Think this falls into documentation again...  From a user perspective, the only interface is the GUI; tinkering with config files I have no problem with *iff* I know they are to be altered.  Hence no migration of .slicing file...

WesGilster, 2015-09-30:
Good point, maybe we need to have a CW file migration document here:
https://github.com/area515/Creation-Workshop-Host/wiki

Step 1:
In CW configuration files are installed with the application in the location where you have installed it.
In CWH configration files are stored in the home directory of the user executing the application.

CW:
On Windows the configuration files are stored in [CWInstallLocation]/Profiles & [CWInstallLocation]/Machines.

CWH:
On Raspbian, it's probably root's home directory: /root/Profiles & /root/Machines 
On Windows, it's something like c:\users\[Username]\Profiles & c:\users\[Username]\Machines
CWH also has the concept of 3D printers which are a combination of a Profile, a Machine and some startup information bundled together to form a .printer file in this directory: c:\users\[Username]\3dPrinters or /root/3dPrinters depending on the OS.

The format of the printer file simply points to the machines and profiles:
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<PrinterConfiguration>
    <MachineConfigurationName>CWH Template Printer</MachineConfigurationName>
    <SlicingProfileName>CWH Template Printer</SlicingProfileName>
    <AutoStart>false</AutoStart>
</PrinterConfiguration>

Step 2:
Expression conversion. This should be automatic, but it's been quite a while since I've tested it, but not all CW expressions are valid FreeMarker expressions. The {} will be automatically added to any CW $ term.

Step 3:
Of course, there is quite a bit more functionality you can use, but at this point you are ready to test, and let me know the problems you face.

WesGilster, 2016-01-18:
This was fixed with the GUI rewrite and the documentation is pretty healthy now:
https://github.com/area515/Creation-Workshop-Host/wiki/Slicing-With-Creation-Workshop-Host

closed 2016-01-18 by WesGilster