area515/Photonic3D #68: About new Raspberry Pi Touchscreen 7"

kitprinter3d, 2015-09-29:
Hi Wes!
Is there a way to get this new screen works with CWH? I have checked with last unstable version but it doesn't recognize the display. 
http://swag.raspberrypi.org/products/raspberry-pi-7-inch-touchscreen-display
I think this upgrade could be crucial for our community. 
Thanks for your amazing job

kloknibor, 2015-09-29:
Wow that would be a nice addon! Thank you can make true standalone printers and at low costs! But I guess there have to be done much work before this can work...

I believe currently CWH doesn't even support reading files from USB to print only wireless and it just makes an server ;)!

Maybe if you truly want this it would be nice to donate one to Wes so he can test and debug it and also as a reward for his hard work :)! (as long as he is into this ofcourse ;))

ergobot, 2015-09-29:
It seems possible but we haven't done any work with it.  This touchscreen
is really new and we don't have one yet.

If anyone sees any examples of using the pi, this touchscreen, and another
screen (using hdmi), post it and I'll take a look.

For reference, the pi display under discussion is here:
https://www.raspberrypi.org/blog/the-eagerly-awaited-raspberry-pi-I display/

More discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?t=120541&p=820953

kitprinter3d, 2015-09-29:
Please check this video:
https://youtu.be/hqdz66Y-H7M?t=1m34s

kloknibor, 2015-09-29:
this was in the post you posted ergobot : 

Dual display usage

It is possible to use both display outputs at the same time, but it does require software to choose the right display. Omxplayer is one application that has been modified to enable secondary display output.

To start displaying a video onto the LCD display (assuming it is the default display) just type:

# omxplayer video.mkv
To start a second video onto the HDMI then:

# omxplayer --display=5 video.mkv
Please note, you may need to increase the amount of memory allocated to the GPU to 128MB if the videos are 1080P, adjust the gpu_mem value in config.txt for this. The Raspberry Pi headline figures are 1080P30 decode, so if you are using two 1080P clips it may not play correctly depending on the complexity of the videos.

Display numbers are:

DISPMANX_ID_FORCE_LCD 4
DISPMANX_ID_FORCE_TV  5
DISPMANX_ID_FORCE_OTHER 6 /* non-default display */

and like you see in the clip dual display is possible :)!

kitprinter3d, 2015-09-29:
I already have mine so please tell me if there is something I can help.

ergobot, 2015-09-29:
omxplayer start
https://github.com/popcornmix/omxplayer/blob/master/omxplayer

With a touchscreen and an hdmi output connected, it might just be a matter
of setting the display.

WesGilster, 2015-09-30:
I would hope that the designers of the DSI interface for the Raspberry pi simply created display drivers for Linux. If so, problem solved, since Linux just passes through the displays to Java as (/display0, /display1)

It also opens a whole range of ideas like automatically starting the the SlicingBrowser and Mask creator directly on the screen. All forms of native Java apps would be at your disposal then. 

It's all pretty much speculation since I don't have one to test out.

My best bet is that we need to address this:
export DISPLAY=:0.0
xinitProcess=`ps -ef | grep grep -v | grep xinit`
if [ -z "${xinitProcess}" ]; then
    echo No X server running, starting and configuring one
    startx &
    xhost +x
fi

to fix kitprinter3d problem...

WesGilster, 2015-09-30:
@kloknibor unfortunately ergobot already had the functionality to load from USB and I took it out. :( I hardcoded to always load from Java's temp directory. If someone is interested, I can put it back in...

WesGilster, 2015-09-30:
@kloknibor I'm on crack, I didn't take the functionality out. Just add this to the config.properties:
uploaddir=/[usb]/[mount]/[point]

WGAndrew, 2016-09-16:
From our experience, X only has affinity to one display at a time on a pi. If you want to use a second display, you've effectively got to circumvent X windows and create and use your own display manager (which is why omxplayer, kivy and a handful of other programs support the touchscreen, and very few others do.)

WesGilster, 2016-09-16:
I've personally used 2 screen sources in X-Windows and Photonic3d on a Linux laptop. I've also used multiple monitors with Photonic3d on a Windows desktop. However, I've never tried the pi so I'll have to trust you.

WesGilster, 2017-06-28:
This was implemented long ago.

closed 2017-06-28 by WesGilster