area515/Photonic3D #71: Can't connect to LittleRP

djambo, 2015-10-06:
I am not sure I am doing something wrong, but no matter how hard I try I can't get CWH on my RPI to drive the controller of the littleRP.

Is there any precise configuration that I am missing?

WesGilster, 2015-10-06:
You'll have to be a bit more specific. Are you having a problem installing CWH, viewing the CWH web page, starting the printer, printing a job or something in between?

djambo, 2015-10-06:
Second sorry for the lack of details. Essentially everything seems to work except the communication between the PI and the LittleRP. I can display the Web UI and send command like "Show Blank to the projector". Also I can "Start the printer", but whenever I try to print something, The log says 1 of 400 slices and nothing happens. Also I tried to control the motor of the printer from the UI but doesn't seems to respond.

WesGilster, 2015-10-06:
What is the file type of what you are printing? Do you see the first image come back to the gui? What version do you have installed? Do you have the CWH version that has the "Send Diagnostics" button? That was a very recent development. If you don't have the button, we'll need to walk through the diagnostic dump manually.

djambo, 2015-10-06:
File type is .cws, I have followed the procedure on the youtube video. I do not see any image coming back to the guy. Right now I am not home but I am sure I have installed the last version after the merge happened 7 days ago. I will be home in a bit and check if the "Send Diagnostics" is there.

WesGilster, 2015-10-06:
The Send Diagnostics is very new, more recent than 7 days. Unfortunately that means we'll need to do take the diagnostic manually, or you'll need to install from the development branch and configure your email settings so that a full diagnostic will automatically be sent to me. 

Is this problem easy to reproduce?

djambo, 2015-10-06:
Yes, it actually happens every time I try to run a print with CWH. I will install the dev branch and try to send you the diagnostics. Where do I edit the email settings and which address should I use?

WesGilster, 2015-10-06:
Cool, that's good it's easy to reproduce. All host settings are in the config.properties file which should be located in /opt/cwh if you'd like to override these settings without having them overwritten on each upgrade you can put a copy of this file inside /root/3dPrinters/config.properties. Be-careful about adding to many properties into this file because it doesn't get overwritten by CWH upgrades. In other words, I wouldn't put anything into this file outside of your email settings. For the "serviceEmailAddresses" setting you can use my google email address, the account name is the same name as my repo.

ergobot, 2015-10-07:
I have a littlerp and have been looking at this issue.  There might be
something going on when creating a new printer then starting.  Will spend
more time looking and document/fix.

ergobot, 2015-10-11:
This is a problem with the baud rate CWH is using to communicate with printer firmware.

By default, CWH baud rate is set to 115200.  To communicate with the littlerp, the baud rate should be 9600.   
SerialManager:188

Steps to replicate
add the following line at SerialManager:189
newComPortSettings.setSpeed(9600); // assuming yours is 115200 
set the speed to something other than what your printer uses

use the gui to create a test printer called testprinter
or
make a GET call with appropriate params (machine/createprinter/{name}/{display}/{port})
localhost:9091/services/machine/createprinter/testprinter/Simulated%20display/%2Fdev%2FttyUSB2

When creating the printer, if CWH uses a different baud rate than the printer firmware, the response from CWH will be slow.  Eventually, when the console shows:
RXTX Warning:  Removing stale lock file. /var/lock/LCK..ttyUSB0
lineRead:null
Firmware Welcome chitchat:

I guess this shows two problems to work on
1.  Finding/using the correct baud rate for printer
2.  Notifying user that baud speed is incorrect

djambo, 2015-10-12:
Glad to know I am able to use the printer in the meantime.
Could you tell me where is the SerialManager file located?

WesGilster, 2015-10-12:
The baud rate isn't modifiable from the GUI yet, and USB comport 
detection only currently works if you have the connection settings for 
your printer correct. You can find the file you are looking for in the 
Machines directory under whichever printer you are starting:

If you are on Windows:
C:\Users\[Your username]\Machines\[YourPrinterName].machine

If you are on the Raspberry pi:
/root/Machines/[YourPrinterName].machine

Here is the format for the file:
<MachineConfig FileVersion="0">
     <MotorsDriverConfig>
         <ComPortSettings>
             <Speed>115200</Speed> <- Change the speed to 9600 for the 
LittleRP

Thanks,
Wes G.

ergobot, 2015-10-12:
Checked some more stuff in to hopefully fix this issue soon (more work needed).  The tldr is gui upgrade, printer configuration added (needs a little more work), exception handling, 

1. The basic gui was reorganized.  The overall layout was moved around to help for a better user experience. start, stop, create, control, etc.  It still needs more work, but it's getting there. 
2. The page "Printer" lets a user create, view, modify printer configuration settings.  When this is all done, it should solve the problem listed here.  It is still a work in progress.
3. Exceptions thrown in the service set the http response code to 500 (instead of bad request), and have json response like {"type": "ExceptionName", "message": "Exception Message"}.  This way, we can pass the exception back to the client in a helpful way.

Here's some screen shots
[image]
[image]
[image]
[image]
[image]
[image]
[image]

djambo, 2015-10-12:
Sweet! this is starting to look really good. I will definitely switch to the new UI.
Also, thank you Wes for the directions.

WesGilster, 2015-10-20:
Sorry everyone, I messed this up pretty bad in the "stable" area515 build. Here's what I've done to fix it, and hopefully the tests will prevent this from happening again in the future. I'll post this in a couple of other places...

Use:
sudo ./startdev.sh 
to get the latest version of CWH from development until we can get the area515 build up-to-date.

Fixed in cwh-0.154
Rewrote EOL parsing and welcome chitchat parsing
Fixed NPE on close issue
Fixed automatic firmware detection on printer start
Fixed tests to perform firmware detection and 3dfirmware serial startup 10 times in a row
Fixed "can't start printer after it started once"
Optimized performance for JSSCCommPort thereby breaking RXTXCommPort
Changed default SerialPort implementation to JSSCCommPort
I'm starting to really dislike the RXTX implementation

WesGilster, 2015-10-27:
Duplicate of #73.

closed 2015-10-27 by WesGilster