area515/Photonic3D #73: Printer not starting in new GUI

kloknibor, 2015-10-18:
Hi, 

I tried to start my printer in the new GUI, my projector gets an black screen... But I can't start any tasks... and CWH doesn't seem to notice there's an printer started... here are the logs : 

https://www.dropbox.com/s/b1q0zvzss6v78kv/logs.rar?dl=0

please tell me how to solve ;)!

WesGilster, 2015-10-18:
There is a new feature that helps us perform this troubleshooting. It takes a live snapshot of what is happening in inside the software(core dump), takes the contents of all important log files, takes the contents of your configuration files, system properties and then zips that information up and sends it to us. All you have to do is setup the email in your config.properties and hit the diagnostics button.

Keep in mind, you shouldn't have to reconfigure your email after every upgrade anymore due to this:
https://github.com/area515/Creation-Workshop-Host/issues/70

ergobot, 2015-10-19:
On first glance of the logs you provided, I think CWH is looping over the first message (firmware chit chat/welcome).  Do you know what firmware your printer is using (marlin, grbl, etc)?

rGlory, 2015-10-19:
I just installed CWH the first time, but have similar issue. I have 3DFracture printer.

First of all after installation on latest raspbian wheezy rxtx library version did not match. I copied file  /usr/share/java/RXTXcomm.jar over /usr/share/java/RXTXcomm.jar and warning was gone. But problem still persisted - in log.out I see a lot of messages:

lineRead:�H�^T�^A�^D(z^B��j
ok

lineRead:�H�^T�^A�^D(z^B��j
ok

I double checked the settings - speed was 115200.  After log gets arround 32M java terminates with out of memory exception:

java.lang.OutOfMemoryError: Java heap space
        at java.util.Arrays.copyOf(Arrays.java:3326)
        at java.lang.AbstractStringBuilder.expandCapacity(AbstractStringBuilder.java:137)
        at java.lang.AbstractStringBuilder.ensureCapacityInternal(AbstractStringBuilder.java:121)
        at java.lang.AbstractStringBuilder.append(AbstractStringBuilder.java:421)
        at java.lang.StringBuilder.append(StringBuilder.java:136)
        at org.area515.resinprinter.gcode.GCodeControl.readUntilOkOrStoppedPrinting(GCodeControl.java:70)
        at org.area515.resinprinter.gcode.GCodeControl.sendGcode(GCodeControl.java:100)
        at org.area515.resinprinter.gcode.GCodeControl.executeSetAbsolutePositioning(GCodeControl.java:120)
        at org.area515.resinprinter.gcode.GCodeControl.readWelcomeChitChat(GCodeControl.java:117)
        at org.area515.resinprinter.printer.Printer.setPrinterFirmwareSerialPort(Printer.java:273)
        at org.area515.resinprinter.serial.SerialManager.assignSerialPortToFirmware(SerialManager.java:200)
        at org.area515.resinprinter.printer.PrinterManager.startPrinter(PrinterManager.java:119)
        at org.area515.resinprinter.services.MachineService.startPrinter(MachineService.java:432)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:483)
        at org.jboss.resteasy.core.MethodInjectorImpl.invoke(MethodInjectorImpl.java:137)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invokeOnTarget(ResourceMethodInvoker.java:296)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:250)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:237)
        at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:356)
        at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:179)

Looks like it accumulates to much data waiting for ok and then crashes. I switched to JSSCCommPort - output looks better:

lineRead:Sprinter^M
2.0 / 17.03.2015^M
start^M
Planner Init^M
Stepper Timer init^M
Free Ram: -28^M
Plan Buffer Size:1280 / 16^M

lineRead:Sprinter^M
2.0 / 17.03.2015^M
start^M
Planner Init^M
Stepper Timer init^M
Free Ram: -28^M
Plan Buffer Size:1280 / 16^M

but still accumulates up to 32M and crashes with out of memory exception. What can I do?

Thanks,
Slava

WesGilster, 2015-10-19:
Interesting, it looks like you and kloknibor are having the same issue after looking at his logs. Can you check to make sure your baud rate is set properly like this topic: #71

...
The baud rate isn't modifiable from the GUI yet, and USB comport
detection only currently works if you have the connection settings for
your printer correct. You can find the file you are looking for in the
Machines directory under whichever printer you are starting:

If you are on Windows:
C:\Users\[Your username]\Machines\[YourPrinterName].machine

If you are on the Raspberry pi:
/root/Machines/[YourPrinterName].machine

Here is the format for the file:
<MachineConfig FileVersion="0">
     <MotorsDriverConfig>
         <ComPortSettings>
             <Speed>115200</Speed> <- Change the speed to 9600 for the
LittleRP

rGlory, 2015-10-19:
Yes 115200 is correct, first of all I validated by running terminal to /dev/ttyUSB0 port. Then when I switched to JSSCCommPort  I did not change speed settings.

kloknibor, 2015-10-20:
@WesGilster We are both using the same printer ;)! it's running an modified version of sprinter... But I remember I tweaked some settings previous time in machines etc... And I just did a fresh install... Wil try to play with that again ;)!

kloknibor, 2015-10-20:
@rGlory try the settings shana/sabotage suggested : https://github.com/sbryan/Draken/tree/master/CWH/configs maybe that is gonna work ;)!

rGlory, 2015-10-20:
@kloknibor I tried it does not work. Looks like it waits for 'ok' literal and it is not produced by that Sprinter firmware.

kitprinter3d, 2015-10-20:
Same problem with GRBL firmware.

kloknibor, 2015-10-20:
@rGlory weird... I got it to work in the past... I must admit I haven't used it that much lately... (It takes me too long to export the slices to the disk...) And shane still seems to have it working :/ On what raspberry pi are you running CWH ? And are you using old or new GUI?

kloknibor, 2015-10-20:
PS : great we're getting more testers around here :)!

rGlory, 2015-10-20:
@kloknibor 
I have that latest Rapberry Pi 2+ (1Gb quad core). I am not sure what is old or new GUI is, just followed instruction to install it and connected through web on port 9091.

Btw when I connect to port using terminal it brings me that welcome message only once - on connection. In log that message there appears many many times until it gets out of memory. Looks like it constantly connects/disconnects or there is some different reason it makes firmware to print that welcome message again and again.

WesGilster, 2015-10-20:
Sorry everyone, I messed this up pretty bad in the "stable" area515 build. Here's what I've done to fix it, and hopefully the tests will prevent this from happening again in the future. I'll post this in a couple of other places...

Use:
sudo ./startdev.sh 
to get the latest version of CWH from development until we can get the area515 build up-to-date.

Fixed in cwh-0.154
Rewrote EOL parsing and welcome chitchat parsing
Fixed NPE on close issue
Fixed automatic firmware detection on printer start
Fixed tests to perform firmware detection and 3dfirmware serial startup 10 times in a row
Fixed "can't start printer after it started once"
Optimized performance for JSSCCommPort thereby breaking RXTXCommPort
Changed default SerialPort implementation to JSSCCommPort
I'm starting to really dislike the RXTX implementation

rGlory, 2015-10-21:
It works now, thanks!

WesGilster, 2015-10-21:
Good news. 

This wasn't really a firmware specific issue, but it would be nice to hear that it's operational under GRBL as well. Any luck KitPrinter3d?

kitprinter3d, 2015-10-21:
I will try later, thank you Wes!

kitprinter3d, 2015-10-25:
It's working. Thanks Wes.

WesGilster, 2015-10-25:
Awesome, that tests GRBL, Sprinter and Marlin. Just curious, are you guys choosing autodetect firmware or specifying the comport(ttyUSB/ttyACM) directly?

rGlory, 2015-10-25:
I am not sure this is the right topic, I noticed that my projector had improper resolution after following installation steps. Disabling overscan in /boot/config.txt fixed the issue. I suggest to put this into installation howto for Pi.

WesGilster, 2015-10-26:
Thanks, I added the line to the Raspberry Pi install doc as you suggested, and was thinking about a way to make this better. However I think you should create a new topic on this. That way we can work on a procedure to use the DisplayManager to set the overscan values in the file automatically. Perhaps it's not that important to make use of every line, but it would be nice to have the conversation in a dedicated thread.

kloknibor, 2015-11-06:
Sorry @WesGilster the old interface works for me... But As soon as I change the resources to resourcesnew it still doesn't work for me... here are the latest logs... fresh raspbian weeze (05-05-2015) and ran startdev.sh (to try if this would help) logs ; https://www.dropbox.com/s/630u8sryvk05rtl/logs%20for%20wes.rar?dl=0

WesGilster, 2015-11-07:
I'm going to need a little more to go on than "it doesn't work". A print doesn't work? The gui doesn't show? You get an error? Also, don't forget to use the diagnostic button for the logs and it gives me tons on information I can't normally get from you. You just need to setup your email in the config.properties and put it in the 3dprinters directory so you never have to setup your email again.

kloknibor, 2015-11-07:
Haha sorry wes, it was kind of an frustrating day ;)! But well what isn't working is that I can connect an printer and the projector turns to blank screen... And I can click on controls but I can't seem to move (or project) anything... And if I click on something else than printers (like print jobs) it jost remains on home screen... And the email diagnostics didn't work either last time... Wil try if this works again later ;)

WesGilster, 2015-11-08:
Sorry Kloknibor. Don't spend too much time in the new gui. 90% of the controls aren't even attached to anything, and printing isn't even close yet. It's really just for getting the look and feel nice and pretty.

Ergobot is also modifying the old gui to make it better as well. Lot's of stuff going on right now, sorry I didn't mention that earlier.

ergobot, 2015-11-08:
Still working on the new gui.

WesGilster, 2015-12-27:
Completed and tested in cwh-0.178 with new GUI.

closed 2015-12-27 by WesGilster