area515/Photonic3D #77: Error: Problem communicating with host printer. (http:500)?

satoer, 2015-11-06:
I’m having a lot of problems with CWH and this error: “Problem communicating with host printer. (http:500)” And I don’t know how to fix it.

What I’ve done:
Installed Wheezy + updated everything and installed CWH. Connected my printer (Draken 1) and it worked fine (controlling Z, and outputting a grid on the display).
I uploaded a CWS file, and halfway uploading CWH seem to have crashed. Reloaded the webpage and “Problem communicating with host printer. (http:500)” error came up. Tried restarting the Raspberry Pi 2, Tried restarting the Printer, power off, power on… but the error remained.

I didn’t know how to reset everything so as last resort I’ve reinstalled everything. But the above error still remains. Even if I unplug the printer the error comes up.

Tried the printer directly in Creation Workshop on the PC, and worked fine, so no problems there.
One other weird thing:
I can’t “create Printer” the “Display’s” selection box is totally empty

Any ideas?

WesGilster, 2015-11-07:
I'll need a diagnostic dump from you:
sudo bash
cd /root/3dPrinters

Edit the following lines in the config.properties file to match your email information(serviceEmailAddresses should be your email as well)
toEmailAddresses=youremailaddress@gmail.com
serviceEmailAddresses=yourprintermanufacturer@gmail.com
username=username
password=password
smtpServer=smtp.googlemail.com
smtpPort=587
mail.smtp.starttls.enable=true

1. Unplug the RPi.
2. Recreate your error.
3. Click the "Send Diagnostics"

You will receive a zipped email that will allow me to fix your problem. After you get this in your email, send it to me.

satoer, 2015-11-07:
Hi Wes,
I’ve edited the Config.properties with my credentials. But there isn’t any button “Send Diagnostics” after power off / on. Checked the config file and can’t find a mistake. But the order of the entries seemed to have changed.

Update, last night I did seem to get a connection somehow, and even printed a part with the Raspberry… but after a reboot (without doing anything with the cables) the error came back :(

WesGilster, 2015-11-08:
It's normal that the entry order changes You don't see the blue "Send Diagnostics" button in the lower right hand corner of the CWH screen shown below? Let's focus on getting the diagnostics to work first, then we'll fix what is causing your print problem. I'm wondering if you are experiencing some sort of timeout. Feel free to take a few diagnostics at different times

[image]

satoer, 2015-11-08:
Sorry, totally overlooked that button. I expected it to be in the error modal.
Anyway, the button gives me an "Host error" So I guess I messed up the smtp credentials. But I can't find any error. Tried a different smtp server from my local provider but keeping that "host error"
Is there somewhere a local log file I can access on the Raspberry?

WesGilster, 2015-11-08:
The logs are stored here:
/opt/cwh

and are called:
log.err
log.out

These will be automatically emailed to me once we get the diagnostics 
working.

Thanks,
Wes G.

satoer, 2015-11-08:
Hi Wes,

After browsing through the past messages I’ve found the mistake. Apparently if you configure raspi-config to start with the startX, it’s going to mess up CWH display’s detection. I’ve reconfigured raspi-config to start with just the terminal. And it worked! The strange thing is that the Raspberry still starts with the GUI. Maybe it’s starting twice if you configure the GUI in the raspi-config?
Maybe a more detailed error modal window would make it easier to troubleshoot errors? “Problem communicating with host printer. (http:500)” does not imply display troubles. My initial suspect was the controller board.

Thanks, Michel.

WesGilster, 2015-11-08:
Yeah, at the moment, CWH doesn't like custom XWindows 
configurations(including the raspi-config option to allow the Raspberry 
Pi to take control). I haven't really looked into it since it plays 
nicely with Ubuntu's Window manager. Anyway, good catch, we should have 
guessed XWindows since the displays were missing. We would have 
eventually caught it once the diagnostics are setup though. I'll make a 
note of this in the Raspberry Pi setup instructions, so that further 
people don't try this setting.

Thanks,
Wes G.

WesGilster, 2016-06-23:
Given that we now have a working Raspberry Pi image, a working and tested install script for the Raspberry Pi, a remote install client for the Raspberry Pi and an install process that works under Linux and tested installations that work under windows; I don't feel like we need to make Photonic3d work under highly customized versions of an OS. That includes installations that use different window managers. Feel free to reopen if anyone disagrees.

closed 2016-06-23 by WesGilster