area515/Photonic3D #84: Detect devices on local network that are capable of running CWH and perform remote install

WesGilster, 2015-12-22:
The CWH client is about to get a bit more functionality:
1. Listen for all 3d printers using DLNA multicast. (Complete)
2. If 1 printer is found, then automatically start that printer in the browser. (Complete)
3. If more than 1 printer is found, then provide a list of printers and their customized names, so that the user can choose which printer to start . (Complete)
4. If no printers are discovered, show all network devices that are capable of running CWH(including all Raspberry Pis)
5. The user chooses the device they want to install CWH and is prompted for a username and password.
6. The user enters the username and password and CWH is installed and shown in the browser.

So to setup a 3d printer(with wifi) from scratch this is the complete set of instructions the end user would have to perform:
1. Plugin all serial cables, hdmi cables, ethernet cable, power cables and sd card(if necessary).
2. Double click on the cwh client, choose the device that you'd like to install CWH onto and enter the username and password for that device. (Example: pi/raspberry)
3. After CWH is installed, it will automatically start in your browser.
4. The user will click the "Settings" tab and select their home Wifi access point, enter the password and click reconfigure.
5. The user will ask to restart the network cable and CWH will close, asking the user to restart with the CWH-client.

WesGilster, 2016-01-21:
The basic functionality is available now, but I need to add a couple more things:
1. "Select your printer" functionality
2. Expand file system from Raspi-Config
3. Enable camera from Raspi-Config

Here is how the process works:
1. The user starts cwh-client.
2. The client is unable to find any printers on the network and shows this:
[image]
3.The user chooses the location they would like to install CWH to and starts receiving feedback as to the install process:
[image]
4. Once CWH has been installed successfully the user is given the option to properly secure SSH. Eventually I'd like to use this same password to enable BasicAuthentication as well. (It would just be a couple of config.properties changes.)
[image]
5. CWH client finds the printer and starts the native browser.
[image]
6. The user chooses their network from the drop down list, enters the password and clicks reconfigure.
[image]
7.The user is asked to unplug the network cable from their device.
[image]
8. Once the cable is unplugged, a network disruption is detected and CWH restarts to complete the setup process (I've found this step is practically unnecessary, but I figured someone's Pi might pickup a whole new IP address than wired so CWH-client would need to find that new address)
[image]

WesGilster, 2016-01-30:
This feature now extends the disk partition and enables the Raspberry Pi camera when installed on the Raspberry Pi. cwhClient-0.212

kloknibor, 2016-03-07:
Works like a charm for me! Might be closed if you ask me

closed 2016-03-07 by WesGilster