area515/Photonic3D #86: Feature request : percentage compleet

kloknibor, 2015-12-30:
Hi,

First I want to say that cwh with new UI runs smooth on my draken ;)! Thanks for all the hardwork!

I'm missing something to know on which layer it is lake : " layer x/xtotal" (and/or an percentage compleet) and maybe an eta just as in creation workshop.

That would be great so you know when your print is finished :)!

Robin

WesGilster, 2015-12-30:
How was the upgrade? You didn't have to do anything, correct? You just restarted the host?
I intentionally left off the progress bar, but I must have left off a little more than I thought. Easy fix.  I'll add those back on my next release which should be in a few days. With the GUI complete you should see some pretty cool options available now that I've been talking about gcode template editing and testing, function graphing, variable size photo/video and that sort of stuff...

WesGilster, 2016-01-03:
Fixed in cwh-0.183.

kloknibor, 2016-01-04:
Hi Wes,

I did an clean install so Sorry I can;t tell you how the upgrade would go... Thanks for the quick fix! 

To be fair I still haven't seen the option you are talking about :$ maybe because I just used an presliced CWS file ? (Because of special Gcode commands and support generation). And I couldn't find my raspberry pi camera anywhere... I hope to find it soon or I will order a new one next month ;)! 

I was running the latest build on a raspberry pi b+ (The raspberri pi 2B was currently in use) and it ran fine ;)! Only the GUI would freeze sometimes, especially on my phone (iphone) and the current slice image took some time to load completly but that will be just the limits of the raspberry pi.

But it really starts to become nice! And since CW is gonna cost like 500 euro I will try you're slicing options ;)! can't wait for your support generation. My guess is that showing a 3D model with the supports is to heavy for an raspberry pi? (So we can check if the support is fine, or maybe you can make such an awesome support generation module that we don't need to :P But currently CW sucks at this and I have to add support manually often). Is there maybe a better (cheap) device to run such functions on? There are so many cheap chinese single board pc's maybe we could use one of those :)

Thanks for all your hard work! And best wishes and happy new year to you ;)!

Robin

WesGilster, 2016-01-05:
Make sure you are using the latest version CWH v0.183. It should show that at the bottom of your screen. I'm not sure which feature you are looking for. If you've got the right version you should see all of those features.
1. Scroll down on the printers screen for GCode editing:
[image]
2. Scroll down on the printers screen and click a test for the JS Calculators:
[image]
3. Click on the currently printed job under print jobs for the other features
[image]

WesGilster, 2016-01-05:
I can understand the unresponsiveness in the iphone browser(those images are huge), but you shouldn't see any kind of freezes with regular browsers. Slow load times, maybe...

WesGilster, 2016-01-05:
Last questions... (Feel free to open separate issues in github for each of your questions next time.) It will be easier to track than this.

Honestly, I feel like we've got a while before we start pushing the power of the quad core Raspberry Pi. On the other hand, I feel like we've grown out of the Single core Raspberry Pi. This is because, unlike other zslicers, I've started parallelizing my algorithm. The reason other slicers are having trouble running on something like a Raspberry Pi is that they waste enormous amounts of time heating up a single core while all of the other cores sit idle. Modern software has to be specifically built to take advantage of multiple CPUs, you don't get to use them for free just because they are available. This is especially true as it relates to algorithms.

The trouble that we could have is web IO. Web servers sometimes have to push quite a bit of data and, in our case, that is in the form of image and video. I can imagine that 3-4 clients could bring the z slicer to it's knees if they are all watching the build and have to be sent images. However, if you don't click on the active print job, you won't be sucking too much bandwidth (you'll just be using a little json). Truthfully I could drop that IO down even further with a couple of client side changes that use websockets exclusively. The server already supports that style of communication.

To you're question about visualizing the model and supports. The good news is that it doesn't have anything to do with Raspberry Pi CPU. Visualization has to do with two things:
1. Web IO for the stl download. (No problems there)
2. Java script browser side 3d modeling software maturity.
I'm not totally convinced on the 3d modeling software yet, I've tried a couple and the stl loading isn't great for huge models. Small models are a dream though and I already support STL highlighted errors so that is going to be really cool when people can visualize that.

Finally on the question of hardware. Linux and the JVM are pretty much ubiquitous so if you find something you like better, let's do it. I probably won't even have to change a line of code shell or java code. To support different hardware, all we'd need to do is find the binary replacements for raspivid/raspistill.

kloknibor, 2016-01-05:
Sorry I will open seperate questions next time :$ Thanks for the effort for answering them ;)! 

I will try to test the new functions this week ;)! And will report to you if I get problems with non ios browsers ;)!

About the visualization I know what you mean it's ran in the browser instead of the raspberry pi. Sometime ago I started with an tool to calculate cost for an print on my website. The tool isn't finished but maybe try if it works for the models you had trouble with : https://drakenprint.nl/print-offerte/ The main code of the viewer is bought, I only edited it to fit my needs but maybe it can help you get an direction you want to go :)! If you need more information about the software, just ask ;)!

WesGilster, 2016-01-06:
Nice app. I tried out and it seemed like it had about the performance 
I've been seeing with other APIs. It was a shade slower than I expected 
since I uploaded a fairly small model. What is the underlying JS API 
that your word press plugin uses? Keep in mind, I don't want the name of 
the Word Press plugin, I want to know the 3d API that the WP plugin uses.

Thanks,
Wes G.

kloknibor, 2016-01-06:
Hi Wes,

It isn't a wordpress app ;)! It's actually loading into an iframe and is an standalone application ;)! I merged some default apps together along with some coding from myself (Just a hobbyist ;) Wish I could do what you do ;p) 

The 3D API is jsc3d ;) The other things used in the app are Jquery, and bootstrap but that won't be interesting I guess ;)!

My guess is that it could be delayed... The app does something kinda stupid... It uploads the stl file to the server and uses the stl to calculate all kinds of parameters like bounding box, layers , max res etc etc and than it sends the same file back to the client browser where it get's downloaded and viewed afterwards. Since my server is located in "Haarlem" the Netherlands there might be some more delaying...!

kloknibor, 2016-01-06:
Here you can find more about JCS3D : https://code.google.com/p/jsc3d/ Including more testers ;)!

kloknibor, 2016-01-06:
THis feature works like a charm again! Can be closed

closed 2016-01-13 by WesGilster