area515/Photonic3D #88: Send a text message when print is complete

WesGilster, 2016-01-01:
You'll use open open auth to sign into your Twillo account and specify the phone number. Then you'll get text messages when the print get's paused due to out of ink or finished. 

The code should work without modification from another project I've been working on...

kloknibor, 2016-01-05:
What would every sms cost?

WesGilster, 2016-01-06:
I use this:
https://www.twilio.com/sms/pricing

kloknibor, 2016-01-08:
How about using some free app?

WesGilster, 2016-01-09:
Email and texting are pretty ubiquitous but there are hundreds of messaging apps on the Internet. If you find a single API that talks to all of them and texts for free, I'll do it. Also keep in mind, developing your own notification is as easy as writing a single method. If you know JavaScript, you could certainly write that. I suppose I could write an upload facility for plugins as well. I'll open a feature for that.