area515/Photonic3D #92: Pause on ignorable errors

WesGilster, 2016-01-03:
Some errors aren't a big deal and shouldn't affect the outcome of the print 3 hours in. In these situations, the printer needs input from the user in order to decide what to do. In these situations the printer will put itself into a paused state with an "ignorable" error. In the event that the user continues the print, the user won't be prompted for that same error. These notifications will get pushed out of the typical notification framework that is already built in...

WesGilster, 2016-01-04:
A couple of examples are:
1. When a PrintFileProcessor doesn't support a feature like buildAreaMM
2. When a PrintMaterialDetector isn't working properly.

kloknibor, 2016-03-07:
this error which the printer returns might be added to the list (photocentric printer) :+1: 
error: Can't execute G28 while door is open at line 'G28'

If the hood is off it refuses to move and gives this error to CWH. It would be best to just pause and give an resume button to continue instead of keep printing and just not moving.

kloknibor, 2016-03-14:
there are several errors... One includes the gcode and I think the other is constant... See here : 

2016-03-14 21:47:55,378 INFO o.a.r.s.Main [main] =================================================================
2016-03-14 21:47:55,384 INFO o.a.r.s.Main [main] =================================================================
2016-03-14 21:47:55,385 INFO o.a.r.s.Main [main] CWH started
2016-03-14 21:47:55,386 INFO o.a.r.s.Main [main] =================================================================
2016-03-14 21:47:55,386 INFO o.a.r.s.Main [main] =================================================================
2016-03-14 21:48:03,378 INFO o.a.r.d.UPNPAdvertiser [main] Relative UPNP root descriptor: /dev/5fba9aca-5489-ea15-0000-00005229b03f/desc
2016-03-14 21:48:33,698 DEBUG o.a.r.p.PrinterManager [qtp30849083-17] Attempting to start printer:Photocentric 10"
2016-03-14 21:48:34,609 INFO o.a.r.d.DisplayManager [qtp30849083-17] Display:X11GraphicsDevice[screen=0] assigned to Printer:Photocentric 10"(printerFirmwareSerialPort:null, projectorSerialPort:null Display::0.0)
2016-03-14 21:48:34,611 DEBUG o.a.r.p.PrinterManager [qtp30849083-17] Assigned display:X11GraphicsDevice[screen=0] to:Photocentric 10"(printerFirmwareSerialPort:null, projectorSerialPort:null Display::0.0)
2016-03-14 21:48:34,630 INFO o.a.r.s.SerialManager [qtp30849083-17] Attempting to assign firmware using serialPort:First available serial port with settings:{Port:First available serial port speed:115200 databits:8 parity:None stopbits:One handshake:None} to printer:Photocentric 10"(printerFirmwareSerialPort:null, projectorSerialPort:null Display::0.0)
2016-03-14 21:48:34,634 INFO o.a.r.s.SerialManager [qtp30849083-17] Attempting to autodetect resources for:PrinterFirmware using serialPort:First available serial port with settings:{Port:First available serial port speed:115200 databits:8 parity:None stopbits:One handshake:None} to printer:Photocentric 10"(printerFirmwareSerialPort:null, projectorSerialPort:null Display::0.0)
2016-03-14 21:48:34,663 DEBUG o.a.r.s.SerialManager [qtp30849083-17] Autodetection trying against serial device:/dev/ttyACM0
2016-03-14 21:48:34,691 INFO o.a.r.s.SerialManager [qtp30849083-17] Resources detected:{comPort:/dev/ttyACM0 model:null settings:{Port:/dev/ttyACM0 speed:115200 databits:8 parity:None stopbits:One handshake:None}}
2016-03-14 21:48:34,807 INFO o.a.r.p.Printer [qtp30849083-17] Firmware serial port set to:/dev/ttyACM0
2016-03-14 21:48:37,020 INFO o.a.r.g.GCodeControl [qtp30849083-17] Write: G91

2016-03-14 21:48:37,328 INFO o.a.r.g.GCodeControl [qtp30849083-17] lineRead: ok G91

2016-03-14 21:48:37,331 INFO o.a.r.p.Printer [qtp30849083-17] Firmware Welcome chitchat:ready
ok G91

2016-03-14 21:48:37,334 INFO o.a.r.s.SerialManager [qtp30849083-17] Completed assignment of firmware using serialPort:/dev/ttyACM0 with settings:{Port:/dev/ttyACM0 speed:115200 databits:8 parity:None stopbits:One handshake:None} to printer:Photocentric 10"(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0)
2016-03-14 21:48:37,336 DEBUG o.a.r.p.PrinterManager [qtp30849083-17] Assigned 3dprinter firmware:First available serial port to:Photocentric 10"(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0)
2016-03-14 21:48:37,338 INFO o.a.r.p.PrinterManager [qtp30849083-17] Printer started:Photocentric 10"(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0)
2016-03-14 21:48:43,939 INFO o.a.r.g.GCodeControl [qtp30849083-43] Write: G91

2016-03-14 21:48:44,242 INFO o.a.r.g.GCodeControl [qtp30849083-43] lineRead: ok G91

2016-03-14 21:48:44,245 INFO o.a.r.g.GCodeControl [qtp30849083-43] Write: G28

2016-03-14 21:48:44,547 INFO o.a.r.g.GCodeControl [qtp30849083-43] lineRead: ok G28

2016-03-14 21:48:47,517 INFO o.a.r.g.GCodeControl [qtp30849083-31] Write: G91

2016-03-14 21:48:47,519 INFO o.a.r.g.GCodeControl [qtp30849083-31] lineRead: error: movement was interrupted because of door opening

2016-03-14 21:48:47,523 INFO o.a.r.g.GCodeControl [qtp30849083-31] lineRead: ok G91

2016-03-14 21:48:47,534 INFO o.a.r.g.GCodeControl [qtp30849083-31] Write: G1 Z0.025 F100.0

2016-03-14 21:48:47,837 INFO o.a.r.g.GCodeControl [qtp30849083-31] lineRead: ok G1 Z0.025 F100.0

2016-03-14 21:48:49,743 INFO o.a.r.g.GCodeControl [qtp30849083-13] Write: G91

2016-03-14 21:48:50,045 INFO o.a.r.g.GCodeControl [qtp30849083-13] lineRead: ok G91

2016-03-14 21:48:50,048 INFO o.a.r.g.GCodeControl [qtp30849083-13] Write: G1 Z1.000 F100.0

2016-03-14 21:48:50,349 INFO o.a.r.g.GCodeControl [qtp30849083-13] lineRead: ok G1 Z1.000 F100.0

2016-03-14 21:48:51,254 INFO o.a.r.g.GCodeControl [qtp30849083-30] Write: G91

2016-03-14 21:48:51,556 INFO o.a.r.g.GCodeControl [qtp30849083-30] lineRead: ok G91

2016-03-14 21:48:51,559 INFO o.a.r.g.GCodeControl [qtp30849083-30] Write: G1 Z1.000 F100.0

2016-03-14 21:48:51,861 INFO o.a.r.g.GCodeControl [qtp30849083-30] lineRead: ok G1 Z1.000 F100.0

2016-03-14 21:49:07,860 DEBUG o.a.r.p.PrinterManager [qtp30849083-30] Attempting to stop printer:Photocentric 10"(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0)
2016-03-14 21:49:07,899 DEBUG o.a.r.p.PrinterManager [qtp30849083-30] Stopped printer:Photocentric 10"(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0)
[... cut ...]
2016-03-14 22:41:34,000 INFO o.a.r.g.GCodeControl [qtp7905343-57] Write: G91

2016-03-14 22:41:34,003 INFO o.a.r.g.GCodeControl [qtp7905343-57] lineRead: error: movement was interrupted because of door opening

2016-03-14 22:41:34,005 INFO o.a.r.g.GCodeControl [qtp7905343-57] lineRead: ok G91

2016-03-14 22:41:34,007 INFO o.a.r.g.GCodeControl [qtp7905343-57] Write: G28 Y

2016-03-14 22:41:34,309 INFO o.a.r.g.GCodeControl [qtp7905343-57] lineRead: error: Can't execute G28 while door is open at line 'G28 Y'

WesGilster, 2016-03-14:
Cool, this will be a nice fix. I'll make an error regular expression matcher that will replay lost gcodes in the event those errors are "experienced". 

However I'm a little confused. I saw in your other stack trace that you had a 2 minute delay every time that you got an error. I can see that's NOT happening (now it's like 2 milliseconds), what are you doing different now?

WesGilster, 2016-03-14:
Adding a couple more questions:

- Where did the 2 minute delay go?
- How is it that you are able to open and close the door in a matter of 5 milliseconds as shown below? 
> 2016-03-14 21:48:47,519 INFO o.a.r.g.GCodeControl [qtp30849083-31] lineRead: error: movement was interrupted because of door opening
> 2016-03-14 21:48:47,523 INFO o.a.r.g.GCodeControl [qtp30849083-31] lineRead: ok G91  

- Could you verify that I actually need to replay the last gcode that was sent (it looks like the firmware is already doing this as shown below)

kloknibor, 2016-03-15:
Things I did different, are that the other error code was from G21 a code the software doesn't know and this returns another error about door opening. 

Uhhh I don't have a clue :P  I ain't a ninja so it will be an software thing. It just handles this error wrong... It could be it sees it isn't a movement and returns it neverthless and executes it. But since G28 Y never returns an ok I suspect he just skips the gcode if sensor is activated... 

What was happening when G21 is send or wrong gcode cwh triggers the 2 minutes waiting time. When hood is open and gcode is send it just dosn't moves and cwh moves a long... It will never move so it doesn't replay itself

This was by the way the complete log from cwh.log

WesGilster, 2016-03-16:
That's kindof funny, I'll keep in mind that you aren't a ninja. :)  I wasn't sure if you were simulating the lid open with a button press or if it was continuing on it's own...

I think I've got enough to go on now...

WesGilster, 2016-04-09:
This seems to be working nicely, but it needs to allow the the print setup to specify which Warnings and Cautions it should pause on.

kloknibor, 2016-04-09:
Sorry still haven't tested... Have some printer location problems :p hoping to test tomorrow

WesGilster, 2016-04-13:
Only responses from the printer that match the regular expression will pause the printer now. You'll probably want a regular expression like the following:
[image]

kloknibor, 2016-04-16:
At first I got some problems, after some logs checking I found out that CW didn't convert my code for Z lift quite well... Will open a seperate issue... When hardcoded this works like a charm! Really great!! Thanks Wes! Only problem is you shouldn't do it while moving... It will stall the printer... But well that's firmware side :/

Great Wes! Thanks... Also can this be set from the printer profile later on? You would make my day :)

WesGilster, 2016-04-17:
Few questions:
1. Shouldn't do what while moving? What is stalling the printer?
2. What more would you want set on the printer profile?

kloknibor, 2016-04-17:
The first question is irrelevent to this bug, it's what I described in #211 so its my fault ;)! 

The second question:
I meant that I could set the .*door.*open.* from the printer config file in /printers . This way it can be set when downloading the printer from github :)

WesGilster, 2016-04-17:
When you set the regex in the "advanced properties" section as the screenshot above shows, it will save it to your printer. Are you trying to say it's not saving?

kloknibor, 2016-04-17:
No it works great Wes no worries! But it would be nice if people wouldn't have to set the value manually but the printer profile would fill it for them :) it's just an extension on what we currently got. What we got now works really great!

WesGilster, 2016-04-17:
I understand that what we have is fine, but I want to understand what you are asking for. That value is already saved and populated by the printer profile. If they choose a MachineConfig where that value is already set it will just work.

kloknibor, 2016-04-17:
ah allright :) What parameter do I need to set for that?

WesGilster, 2016-04-17:
In Eclipse, ctrl-shift-t -> MachineConfig -> @XmlElement(name="PauseOnPrinterResponseRegEx")

kloknibor, 2016-06-21:
Works great! Can be closed

WesGilster, 2016-06-23:
Implemented long ago. Closing.

closed 2016-06-23 by WesGilster