area515/Photonic3D #94: Failure emailing log bundle.

kloknibor, 2016-01-05:
Hi,

I'm having an error so I tried to send a .log bundle to myself but it's failing... I guess because I try to use an "@msn.com" account So I set the SMTP host to smtp.live.com but I guess there also needs to be a changed incoming port, is that correct? Or is it something else?

Robin

WesGilster, 2016-01-06:
You'll need to look at the logs and send me the stack trace manually. 
Try these in this order:
1. Does MSN use port 587?
2. Does MSN use TLS?
3. If MSN uses TLS what is the port number?

If it's none of these, recheck your usernames and passwords. I use this 
function quite a bit. In the mean time, I'll rename this bug to give 
better feedback as to why the email failed.

Thanks,
Wes G.

kloknibor, 2016-01-06:
According to this page : https://support.microsoft.com/en-us/kb/287604 

1 : Yes it does use port 587
2 : Yes MSN uses TLS
3 : Port 25 by default but port 587 is acceptable too if I understand it correct

will send you the files later

WesGilster, 2016-01-07:
Much better error messages when email fails to send in cwh-0.185

kloknibor, 2016-01-08:
Gonna try it tonight!

kloknibor, 2016-01-08:
I get the following error : Failure emailing log bundle: 501 5.5.4 Invalid Email address

kloknibor, 2016-01-08:
(trying on windows now quick)

kloknibor, 2016-01-08:
Edit : I tried an gmail account and it blocked the mail because an app which doesn't apply to modern security tried to log in...  No option in the mail to give it acces anyway...

kloknibor, 2016-01-08:
If more people experience this than log in to your google account , go here : https://www.google.com/settings/security/lesssecureapps and click enable ;)! Now Gmail works but msn still doens't for some reason...

WesGilster, 2016-01-09:
I assume that it uses oath now. I'll open a feature request for that.

closed 2016-01-18 by WesGilster