area515/Photonic3D #97: User permissions

WesGilster, 2016-01-09:
Cwh provides unprecedented control over how your printer prints. But this control needs to be capped with user permissions. Basic auth is already available so it's not really a security issue but certain functions need to be further secured per user/group.

WesGilster, 2016-06-23:
My last commit should be a pretty large step forward in the area of User management, authorization and authentication. When you guys get a chance I'd like you to try it out. I still need to add the user management functions through the GUI, but you can get a user migrated to the new mechanism by adding a user to the config.properties in the 3dprinters directory:

> useSSL=true
> CWH3DPrinterRealm.clientUsername=admin
> CWH3DPrinterRealm.clientPassword=password

Don't add the user to the regular config.properties or you'll go through pain and agony. Here are the current features:

1. Totally pluggable user management features: It allows the developer to implement what it means to "CRUD" a user by implementing the org.area515.resinprinter.security.UserManagementFeature interface.
2. Jetty based authentication mechanism: This allows the developer to define how a user logs into Photonic3d. I reused the org.eclipse.jetty.security.LoginService interface so that we can reuse all of the login services available in Jetty, but this choice does lock us into Jetty a bit further.
3. KeystoreLoginService implementation of user management: When users are created, an encryption and a signing communications key is created and cryptographically stored against their login password in a Java keystore. The password is irretrievable.
4. Immediately revocable permissions: When a user is deleted by another user, that user is immediately logged out of the application and doesn't have the ability to make further restful calls.
5. Fine grained annotated role based authorization: Each function in the restful interface can be assigned the specific roles that a user requires to execute it. Currently these functions are coarsely set at the Service level.

This is really just a setup for the next big user trust and communications plugin I have planned. This phase will ultimately bring together the entire community of Photonic3d users in a pretty big way.

kloknibor, 2016-06-24:
I will try to test it on sunday!

WesGilster, 2016-06-27:
I just checked in the next phase of the Photonic3d User trust and communication protocol and a few tests(including a couple of exploit attempts). This defines the P2P secure messaging that we'll be using for remote communications. Hopefully this should give the community a chance to pick apart my crypto algorithm a bit before we have a prod version ready.

I realize the choice of X509 over PGP for P2P is an odd choice, but I did it to keep bouncy castle dependencies out of the build for now. If we get a huge set of requests for a web-of-trust style of pki, I really don't think it would be tough to generify the code to PGP. I've tried to keep it fairly generic for the possibility of that extension anyway.

My communications protocol is also designed to treat our own eventual rendezvous server as an adversarial man in the middle. This is unlike most "cloud" style storage or social frameworks which actively peer into every message sent and "trust" the central servers. In this model, only the end user on the other side is trusted.

jmkao, 2016-06-27:
How do you think that this functionality will be used so that we can try out those scenarios?

WesGilster, 2016-06-27:
There are two phases that are done. You can actually test the first phase by inserting the three properties above to create a new user. This will turn on basic auth and allow you to authenticate and authorize your user to all restful methods. If you'd like to experiment with changing the permissions of a restful method, you can do that at the method or at the class level with the @RolesAllowed annotation. Then you'll get to see some authorization in action. Until I stop hardcoding user roles, all users are authorized to all methods.

The second phase brought in the concept of remote users. Remote users can be assigned roles as well, and in the beginning will only "listen" to web socket events that the printer pushes. The only way to test the "Message" protocol at the moment is to write a quick program like the tests are doing now. As I mentioned in the checkin, I'm currently using X509 for my PKI implementation, but could reimplement with PGP if there is any sort of push. What I'm hoping is that people now hold their social network account in high enough regard that public key distribution through that channel is acceptable. Of course I can't facilitate that process because it would taint the already scketchy out-of-band requirements, but you get the idea...

Hmm, looks like Facebook is going down the pgp route, so I may have to implement with PGP sooner than I thought:
http://thenextweb.com/facebook/2015/06/01/facebook-now-lets-you-put-a-pgp-key-on-your-profile-and-uses-it-to-encrypt-notifications/#gref

WesGilster, 2018-01-21:
Users and Roles for Photonic functionality have been a part of Photonic3D for quite some time. I left this bug open because I was thinking about utilizing PGP for the encryption, but there really isn't any point in that.

closed 2018-01-21 by WesGilster