area515/Photonic3D #98: Print text files

WesGilster, 2016-01-09:
This is the next thing I want to print with abstract printing. Choose the font and size in print setup and upload like usual. Regular is fonts supported by default.

WesGilster, 2016-02-01:
This is complete in the dev version cwh-0.215. You can pick the font name from any available system fonts and font size on the printer setup screen now.

I've also rearranged the printers screen, it still doesn't look great, but it uses two columns now.

WesGilster, 2016-03-06:
It was my daughters birthday weekend so I didn't do anything except some print testing. These are my first tests for text file prints.

[image]

kloknibor, 2016-03-06:
Looking nice Wes! And happy birthday to your daughter :)!

Verstuurd vanaf mijn iPhone

closed 2016-03-11 by WesGilster