area515/Photonic3D #99: SliceBrowser quits at launch

jmkao, 2016-01-10:
When launching the Slice Browser (e.g. "java -cp lib/*:. org.area515.resinprinter.slice.SliceBrowser"), a message window appears with the message: "File not found:C:\Users\wgilster\Documents\ArduinoMegaEnclosureBottom.stl".

After clicking the "OK" button, some numbers are printed to the console and the program exits.

If I create an STL file (I am running on Linux) named 'C:\Users\wgilster\Documents\ArduinoMegaEnclosureBottom.stl', then the program prints some numbers to the console and then exits without opening an windows.

WesGilster, 2016-01-10:
Sorry about that. I must have turned off the GUI at some point. Turned it back on in: cwh-0.186

jmkao, 2016-01-11:
The UI comes up now. Thanks!

closed 2016-01-11 by jmkao