area515/Photonic3D #101: Returning Projector Serial Port to Unconfigured Leaves Non-Working Configuration Behind

jmkao, 2016-01-11:
I am currently working on getting basic functionality working on the Kudo3D Titan 1 printer (Marlin based with only a Z axis and no endstops).

If I do not set a value for Projector Serial Port when creating a new printer, I am able to start the printer and controll the Z-axis through controls without any problem.

[image]

If I set a value for the Projector Serial Port and then change it back to "Select a Serial Port", the `<ComPortSettings>` in the machine file is left behind with the values, only removing `<PortName>`:

```xml
        <ComPortSettings>
            <Speed>0</Speed>
            <Databits>0</Databits>
        </ComPortSettings>
```

With these values present, I am unable to Start the printer. The log.err shows the following (which interestingly correctly locates /dev/ttyUSB1 as my projector):

```
jssc.SerialPortException: Port name - /dev/ttyUSB1; Method name - closePort(); Exception type - Port not opened.
        at jssc.SerialPort.checkPortOpened(SerialPort.java:878)
        at jssc.SerialPort.closePort(SerialPort.java:1088)
        at org.area515.resinprinter.serial.JSSCCommPort.close(JSSCCommPort.java:62)
        at org.area515.resinprinter.serial.SerialManager.is3dFirmware(SerialManager.java:90)
        at org.area515.resinprinter.serial.SerialManager.detectResourcesAndAssignPort(SerialManager.java:113)
        at org.area515.resinprinter.serial.SerialManager.assignSerialPortToProjector(SerialManager.java:150)
        at org.area515.resinprinter.printer.PrinterManager.startPrinter(PrinterManager.java:120)
        at org.area515.resinprinter.services.PrinterService.startPrinter(PrinterService.java:190)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:483)
        at org.jboss.resteasy.core.MethodInjectorImpl.invoke(MethodInjectorImpl.java:137)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invokeOnTarget(ResourceMethodInvoker.java:296)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:250)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:237)
        at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:356)
        at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:179)
        at org.jboss.resteasy.plugins.server.servlet.ServletContainerDispatcher.service(ServletContainerDispatcher.java:220)
        at org.jboss.resteasy.plugins.server.servlet.HttpServletDispatcher.service(HttpServletDispatcher.java:56)
        at org.jboss.resteasy.plugins.server.servlet.HttpServletDispatcher.service(HttpServletDispatcher.java:51)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
        at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:769)
        at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1667)
        at org.eclipse.jetty.websocket.server.WebSocketUpgradeFilter.doFilter(WebSocketUpgradeFilter.java:172)
        at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1650)
        at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:583)
        at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:221)
        at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1125)
        at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515)
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
        at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1059)
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
        at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:52)
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97)
        at org.eclipse.jetty.server.Server.handle(Server.java:497)
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:311)
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:248)
        at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:610)
        at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:539)
        at java.lang.Thread.run(Thread.java:744)
java.lang.NullPointerException
        at org.area515.resinprinter.serial.JSSCCommPort.open(JSSCCommPort.java:23)
        at org.area515.resinprinter.serial.SerialManager.is3dFirmware(SerialManager.java:69)
        at org.area515.resinprinter.serial.SerialManager.detectResourcesAndAssignPort(SerialManager.java:113)
        at org.area515.resinprinter.serial.SerialManager.assignSerialPortToProjector(SerialManager.java:150)
        at org.area515.resinprinter.printer.PrinterManager.startPrinter(PrinterManager.java:120)
        at org.area515.resinprinter.services.PrinterService.startPrinter(PrinterService.java:190)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[... 64 lines cut ...]
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:248)
        at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:610)
        at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:539)
        at java.lang.Thread.run(Thread.java:744)
```

In order to restore functionality to that printer, I have to manually go into the machine configuration and remove the `<ComPortSettings>` element completely from the MonitorDriverConfig to return it to the state when it was originally created, looking like:

```xml
    <MonitorDriverConfig>
        <DLP_X_Res>1824.0</DLP_X_Res>
        <DLP_Y_Res>984.0</DLP_Y_Res>
        <MonitorID>DISPLAY1</MonitorID>
        <OSMonitorID>:0.0</OSMonitorID>
        <DisplayCommEnabled>false</DisplayCommEnabled>
        <MonitorTop>0</MonitorTop>
        <MonitorLeft>0</MonitorLeft>
        <MonitorRight>0</MonitorRight>
        <MonitorBottom>0</MonitorBottom>
        <UseMask>false</UseMask>
    </MonitorDriverConfig>
```

After doing this, my printer will Start again.

WesGilster, 2016-01-11:
Looks like my GUI got a little overzealous on setting up the MonitoryDriver. The null pointer is the missing  section you had to rebuild. Thanks for listing the steps you took to recreate the issue. I'll try to fix this tomorrow.

WesGilster, 2016-01-12:
I took a look at what's going on and I decided I'll build a comport editing directive and use that to allow projector and firmware serial port settings editing.

WesGilster, 2016-01-13:
You should be able to completely fill out the settings for your comports by using the "advanced" checkbox as shown below. Fixed in development version: cwh-0.1.90. I'll promote to prod if it works well enough for you...

[image]

jmkao, 2016-01-14:
This doesn't seem to be working. No matter what I have Projector Serial Port set to, it seems to always  do projector autodetection. Once you activate projector autodetection for the config, there seems to be now way to disable it from the UI. The only way to go back to no projector detection is to go into the machine config and delete the `<ComPortSettings>` element entirely.

LogBundle is below:
[LogBundle.zip](https://github.com/area515/Creation-Workshop-Host/files/89766/LogBundle.zip)

WesGilster, 2016-01-17:
I did fix an NPE during the projector autodetection process that was preventing a good detection. However ComPortSettings have always been managed properly by the GUI, from what I've been able to tell. Selecting "Select a projector serial port..." in the list will disable the projector functions. Let me know if you can still recreate this issue after cwh-0.200.

jmkao, 2016-01-17:
I can no longer reproduce this issue. I can successfully "de-configure" the projector from a printer. Looks fixed to me. :)

closed 2016-01-17 by jmkao